introlab/rtabmap_ros #1198: obstacles_detection node: Unused parameters Grid/RangeMax and Grid/RangeMin

Alessio-Parmeggiani, 2024-08-16:
Hi, I noticed that the obstacles_detection node from rtabmap_util (used in ROS2 Humble) has some unused parameters, in particular I noticed that Grid/rangeMax and Grid/RangeMin do not change the resulting pointclouds.

In the following image RangeMax is set to 3 and rangeMin is set to 1. Each cell correspond to 1 meter
[image]

[Looking at the code ](https://github.com/introlab/rtabmap_ros/blob/a97efff760720132ced2a607f384dc5d28a0d296/rtabmap_util/src/nodelets/obstacles_detection.cpp#L210-L218 )it seems that they are not considered at all. 
If these parameters are not used they should not be exposed as parameters of the node, is this the intended behavior?

It would be possible to add the feature of considering a minimum and maximum range for the input and output pointclouds?

matlabbe, 2024-08-30:
Humm, they should be used. How did you set the parameters? The ROS2 code of that node is currently on ros2 branch here: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_util/src/nodelets/obstacles_detection.cpp

```
ros2 run rtabmap_util obstacles_detection --ros-args -p Grid/RangeMin:="'1'" -p Grid/RangeMax:="'9'"
[INFO] [1724980664.169049303] [obstacles_detection]: obstacles_detection: Setting parameter "Grid/RangeMax"="9"
[INFO] [1724980664.169197474] [obstacles_detection]: obstacles_detection: Setting parameter "Grid/RangeMin"="1"
```

Alessio-Parmeggiani, 2024-08-30:
Hi! 
I passed the parameters from launch file and I see that they are set up properly both in the prints of the node and by doing: `ros2 param dump obstacle_detection_node`. 
They do not seem to be considered during the computations, as the resulting pointclouds with ground and obstacle points will consider all the points of the input pointcloud ignoring both the minimum and the maximum range, as in the image I posted before

matlabbe, 2024-09-06:
These two parameters were indeed ignored. It is fixed in the commit above. Thanks for reporting!

commit 9deb8a2a (2024-09-06): Fixed #1198
closed 2024-09-06 by matlabbe