introlab/rtabmap_ros #1200: ROS2 humble on ARM, OpenCV charuco.hpp not found error

Al-khattab, 2024-08-22:
Hi, I am trying to build rtabmap_ros from source on a jetson board that is running Ubuntu22.04 and jetpack 6.0 with ROS2 humble, 
I am following the build steps provided in the ros2 branch and when I build I get the following error 

```
/ros2_ws/src/rtabmap/guilib/src/../include/rtabmap/gui/CalibrationDialog.h:43:10: fatal error: opencv2/aruco/charuco.hpp: No such file or directory
   43 | #include <opencv2/aruco/charuco.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/build.make:164: guilib/src/CMakeFiles/rtabmap_gui.dir/PreferencesDialog.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:923: guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2
---
Failed   <<< rtabmap [5min 8s, exited with code 2]
```

matlabbe, 2024-08-23:
Could be a logic error in this code: 
https://github.com/introlab/rtabmap/blob/6ce85b4e55adc2903a6f08dd25fa0c894b34c9cd/guilib/include/rtabmap/gui/CalibrationDialog.h#L42-L45

Which OpenCV version is used? Can you do `make` under `/ros2_ws/build/rtabmap`?

Al-khattab, 2024-08-25:
as far as I understand, the package will install the dependencies, correct? so I do not need to install OpenCV manually?

also after running the command `make VERBOSE=1` under `/ros2_ws/build/rtabmap`
this is the output 

/usr/bin/cmake -S/home/jetson/ros2_ws/src/rtabmap -B/home/jetson/ros2_ws/build/rtabmap --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/jetson/ros2_ws/build/rtabmap/CMakeFiles /home/jetson/ros2_ws/build/rtabmap//CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
make -f utilite/src/CMakeFiles/rtabmap_utilite.dir/build.make utilite/src/CMakeFiles/rtabmap_utilite.dir/depend
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
cd /home/jetson/ros2_ws/build/rtabmap && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jetson/ros2_ws/src/rtabmap /home/jetson/ros2_ws/src/rtabmap/utilite/src /home/jetson/ros2_ws/build/rtabmap /home/jetson/ros2_ws/build/rtabmap/utilite/src /home/jetson/ros2_ws/build/rtabmap/utilite/src/CMakeFiles/rtabmap_utilite.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
make -f utilite/src/CMakeFiles/rtabmap_utilite.dir/build.make utilite/src/CMakeFiles/rtabmap_utilite.dir/build
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
make[2]: Nothing to be done for 'utilite/src/CMakeFiles/rtabmap_utilite.dir/build'.
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
[ 5%] Built target rtabmap_utilite
make -f utilite/resource_generator/CMakeFiles/res_tool.dir/build.make utilite/resource_generator/CMakeFiles/res_tool.dir/depend
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
cd /home/jetson/ros2_ws/build/rtabmap && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jetson/ros2_ws/src/rtabmap /home/jetson/ros2_ws/src/rtabmap/utilite/resource_generator /home/jetson/ros2_ws/build/rtabmap /home/jetson/ros2_ws/build/rtabmap/utilite/resource_generator /home/jetson/ros2_ws/build/rtabmap/utilite/resource_generator/CMakeFiles/res_tool.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
make -f utilite/resource_generator/CMakeFiles/res_tool.dir/build.make utilite/resource_generator/CMakeFiles/res_tool.dir/build
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
make[2]: Nothing to be done for 'utilite/resource_generator/CMakeFiles/res_tool.dir/build'.
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
[ 6%] Built target res_tool
make -f corelib/src/CMakeFiles/rtabmap_core.dir/build.make corelib/src/CMakeFiles/rtabmap_core.dir/depend
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
cd /home/jetson/ros2_ws/build/rtabmap && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jetson/ros2_ws/src/rtabmap /home/jetson/ros2_ws/src/rtabmap/corelib/src /home/jetson/ros2_ws/build/rtabmap /home/jetson/ros2_ws/build/rtabmap/corelib/src /home/jetson/ros2_ws/build/rtabmap/corelib/src/CMakeFiles/rtabmap_core.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
make -f corelib/src/CMakeFiles/rtabmap_core.dir/build.make corelib/src/CMakeFiles/rtabmap_core.dir/build
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
make[2]: Nothing to be done for 'corelib/src/CMakeFiles/rtabmap_core.dir/build'.
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
[ 57%] Built target rtabmap_core
make -f guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/build.make guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/depend
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
cd /home/jetson/ros2_ws/build/rtabmap && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jetson/ros2_ws/src/rtabmap /home/jetson/ros2_ws/src/rtabmap/guilib/src /home/jetson/ros2_ws/build/rtabmap /home/jetson/ros2_ws/build/rtabmap/guilib/src /home/jetson/ros2_ws/build/rtabmap/guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
make -f guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/build.make guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/build
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
[ 57%] Automatic MOC and UIC for target rtabmap_gui
cd /home/jetson/ros2_ws/build/rtabmap/guilib/src && /usr/bin/cmake -E cmake_autogen /home/jetson/ros2_ws/build/rtabmap/guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/AutogenInfo.json Release
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
[ 57%] Built target rtabmap_gui_autogen
make -f guilib/src/CMakeFiles/rtabmap_gui.dir/build.make guilib/src/CMakeFiles/rtabmap_gui.dir/depend
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
cd /home/jetson/ros2_ws/build/rtabmap && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jetson/ros2_ws/src/rtabmap /home/jetson/ros2_ws/src/rtabmap/guilib/src /home/jetson/ros2_ws/build/rtabmap /home/jetson/ros2_ws/build/rtabmap/guilib/src /home/jetson/ros2_ws/build/rtabmap/guilib/src/CMakeFiles/rtabmap_gui.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
make -f guilib/src/CMakeFiles/rtabmap_gui.dir/build.make guilib/src/CMakeFiles/rtabmap_gui.dir/build
make[2]: Entering directory '/home/jetson/ros2_ws/build/rtabmap'
[ 58%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/rtabmap_gui_autogen/mocs_compilation.cpp.o
[... cut ...]
/home/jetson/ros2_ws/build/rtabmap/guilib/src/rtabmap_gui_autogen/NBNA5662NX/../../../../../../src/rtabmap/guilib/include/rtabmap/gui/CalibrationDialog.h:43:10: fatal error: opencv2/aruco/charuco.hpp: No such file or directory
43 | #include <opencv2/aruco/charuco.hpp>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/build.make:136: guilib/src/CMakeFiles/rtabmap_gui.dir/rtabmap_gui_autogen/mocs_compilation.cpp.o] Error 1
make[2]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'
make[1]: *** [CMakeFiles/Makefile2:923: guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
make[1]: Leaving directory '/home/jetson/ros2_ws/build/rtabmap'

matlabbe, 2024-08-26:
It seems jetpack6 is using OpenCV 4.8 unlike Ubuntu 22.04 Desktop having OpenCV 4.5.4 and including aruco module. Since OpenCV 4.7.0, charuco module [has been moved](https://github.com/opencv/opencv/pull/22986) from opencv_contrib to opencv repo.

I'll make a local build with latest opencv to fix that issue.

closed 2024-08-26 by matlabbe

matlabbe, 2024-08-26:
The commit above should fix the issue. You will need to update `rtabmap` code to latest.