introlab/rtabmap_ros #1204: Could NOT find BLAS (missing: BLAS_LIBRARIES) error during colcon build

Anchor1566, 2024-09-01:
I encountered an issue when building the rtabmap and rtabmap_ros repositories in my ROS2 workspace. The build process fails with the error message Could NOT find BLAS (missing: BLAS_LIBRARIES).
The first and initial build was successful. but then I do some additional operations as(i am not sure if these will affect the build):
1.I updated the GCC version to 12.3
2. I included the ZED library in the src directory
I'm not having much luck trying to find out how to fix this. Can you break it down real simply for me?
Steps to reproduce:
```
cd ~/ros2_ws
git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
rosdep update && rosdep install --from-paths src --ignore-src -r -y
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```
Error Message
```
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find BLAS (missing: BLAS_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.22/Modules/FindBLAS.cmake:1337 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:36 (find_package)
  CMakeLists.txt:601 (find_package)
  ```
OS: ubuntu 22.04
ROS2 Distribution: ros2 humble

matlabbe, 2024-09-05:
Similar issue here https://github.com/stella-cv/stella_vslam_ros/discussions/174

What is your machine? jetpack? `libgdal.so` is not used by rtabmap directly, it is coming from one of the dependence.

One person suggested [here](https://stackoverflow.com/questions/69255382/curl-symbol-lookup-error-curl-undefined-symbol-curl-url-cleanup) to do:
```
sudo ldconfig
```

Anchor1566, 2024-09-06:
> Similar issue here [stella-cv/stella_vslam_ros#174](https://github.com/stella-cv/stella_vslam_ros/discussions/174)
> 
> What is your machine? jetpack? `libgdal.so` is not used by rtabmap directly, it is coming from one of the dependence.
> 
> One person suggested [here](https://stackoverflow.com/questions/69255382/curl-symbol-lookup-error-curl-undefined-symbol-curl-url-cleanup) to do:
> 
> ```
> sudo ldconfig
> ```

Thank you for your assistance and for taking the time to help resolve this issue!
 I'm pleased to report that the problem has been resolved on my end. I was using Ubuntu 22.04  but experiencing errors when attempting to use colcon build with a conda virtual environment. However, after deactivating the conda environment (`conda deactivate`), the compilation process proceeded successfully.

closed 2024-09-06 by matlabbe

jchai01, 2025-06-19:
After deactivating conda, I had to run `rm -rf build install log` for it to work