introlab/rtabmap_ros #1212: RTAB-Map/install/rtabmap/lib/librtabmap_core.so.0.21: undefined symbol: _ZNK5gtsam15NonlinearFactor5rekeyERKSt6vectorImN3tbb6detail2d113tbb_allocatorImEEE

mintrhey, 2024-09-19:
ROS2 Version: Humble
RTAB-Map Version: RTAB-Map ROS2 Latest

I can launch the package without a problem before but I stopped since I need to do some other experimentation and when I run it again it will return these kind of errors. Cannot launch properly all the executable and launch files of the package. The errors persist even the package is installed from binary or source. Is there a way to resolve this?

[image]

matlabbe, 2024-09-22:
It looks like rtabmap and gtsam binaries versions diverged. That can happen when one package has been upgraded with apt-get while the other didn't (but it should have). Try:
```
sudo apt update && sudo apt upgrade
```

mintrhey, 2024-09-24:
Good day! Thank you for the reply. I did what you said but it didn't work but you are right that rtabmap and gtsam binaries versions diverged. I had to remove the installed libraries related to the package I want to use and that includes gtsam but still it didn't work so I had to purge it. There are lots of affected packages and I installed it again including the rtabmap. It does work now however I can't use the package related to gtsam since the error will show again. If you have a recommendation of a way to use both packages without a problem, I would highly appreciate it! Besides that, thank you again.

matlabbe, 2024-09-24:
Oh ok , so there is another package building gtsam internally and may install those gtsam libraries on the PATH or LD_LIBRARY_PATH. If you are using rtabmap binaries, you would have to make sure that when it launches, the other gtsam libraries of the other package are not shadowing the gtsam ros binaries. For example, if the other package installs gtsam in your local ros workspace, you could just do `source /opt/ros/humble/setup.bash` (to remove your workspace on the PATH) before doing `ros2 launch rtabmap_launch ...`.

Another way could be to uninstall gtsam binaries, then rebuild rtabmap from source with the new gtsam installed in your workspace.

 cheers,
Mathieu

mintrhey, 2024-09-26:
Oh okay. Noted. Thank you so much! 

Lucy