introlab/rtabmap_ros #1213: Error running on ROS2 Jazzy: Did not receive data since 5 seconds

michelgokan, 2024-09-26:
I'm trying to run rtabmap_ros with ROS2 Jazzy using my Intel RealSense D435i following [this example](https://raw.githubusercontent.com/introlab/rtabmap_ros/refs/heads/ros2/rtabmap_examples/launch/realsense_d435i_color.launch.py). 

Here is the command:

```
ros2 launch rtabmap_examples realsense_d435i_color.launch.py
```

Here is the output:

```
[INFO] [launch]: All log files can be found below /home/m/.ros/log/2024-09-26-01-50-14-131146-m-69364
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [69367]
[INFO] [rgbd_odometry-2]: process started with pid [69368]
[INFO] [rtabmap-3]: process started with pid [69369]
[INFO] [rtabmap_viz-4]: process started with pid [69370]
[INFO] [imu_filter_madgwick_node-5]: process started with pid [69371]
[INFO] [static_transform_publisher-6]: process started with pid [69372]
[static_transform_publisher-6] [WARN] [1727308214.669830541] []: Old-style arguments are deprecated; see --help for new-style arguments
[imu_filter_madgwick_node-5] [INFO] [1727308214.861595578] [imu_filter_madgwick]: Starting ImuFilter
[imu_filter_madgwick_node-5] [INFO] [1727308214.874984366] [imu_filter_madgwick]: Using dt computed from message headers
[imu_filter_madgwick_node-5] [INFO] [1727308214.875127291] [imu_filter_madgwick]: The gravity vector is kept in the IMU message.
[imu_filter_madgwick_node-5] [INFO] [1727308214.875430455] [imu_filter_madgwick]: Imu filter gain set to 0.100000
[imu_filter_madgwick_node-5] [INFO] [1727308214.875479936] [imu_filter_madgwick]: Gyro drift bias set to 0.000000
[imu_filter_madgwick_node-5] [INFO] [1727308214.875508232] [imu_filter_madgwick]: Magnetometer bias values: 0.000000 0.000000 0.000000
[static_transform_publisher-6] [INFO] [1727308215.160699086] [static_transform_publisher_nF54VqRF0ny0kbjh]: Spinning until stopped - publishing transform
[static_transform_publisher-6] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-6] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-6] from 'camera_gyro_optical_frame' to 'camera_imu_optical_frame'
[realsense2_camera_node-1] [INFO] [1727308215.504176349] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1727308215.504488383] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1727308215.504584604] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1727308216.511496749] [camera.camera]: Device with serial number 233522073357 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1727308216.511697988] [camera.camera]: Device with physical ID /sys/devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb5/5-1/5-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1727308216.511760117] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1727308216.512053670] [camera.camera]: Device with port number 5-1 was found.
[realsense2_camera_node-1] [INFO] [1727308216.513883079] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1727308216.514160632] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1727308216.516946941] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1727308216.517895636] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1727308216.517961136] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1727308216.517994302] [camera.camera]: Device Serial No: 233522073357
[realsense2_camera_node-1] [INFO] [1727308216.518027580] [camera.camera]: Device physical port: /sys/devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb5/5-1/5-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1727308216.518074764] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1727308216.518122468] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1727308216.518156930] [camera.camera]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1727308216.673206886] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1727308216.677055666] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1727308216.770241579] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1727308216.787870330] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1727308216.791285262] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1727308216.860278020] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1727308216.860462018] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1727308216.860531480] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1727308216.947376560] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1727308216.984194067] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1727308217.039017339] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1727308217.091579094] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1]  26/09 01:50:17,118 WARNING [281473152245824] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1727308217.153350082] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1727308217.168177191] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[realsense2_camera_node-1] [INFO] [1727308217.169664511] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-1] [INFO] [1727308217.251956058] [camera.camera]: RealSense Node Is Up!
[rtabmap-3] [INFO] [1727308217.959582529] [rtabmap]: rtabmap(maps): latch                      = true
[rtabmap-3] [INFO] [1727308217.966129878] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-3] [INFO] [1727308217.966254618] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-3] [INFO] [1727308217.966297247] [rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-3] [INFO] [1727308217.966326950] [rtabmap]: rtabmap(maps): map_always_update          = false
[... 254 lines cut ...]
/rtabmap/republish_node_data
/tf
/tf_static
/user_data_async
```

Here is the output of `hz` command:

```
$ ros2 topic hz /camera/camera/color/image_raw
average rate: 14.192
	min: 0.011s max: 615s std dev: 0.07643s window: 250
```

```
$ ros2 topic hz /camera/camera/color/camera_info
average rate: 17.491
	min: 0.004s max: 0.161s std dev: 0.06321s window: 21
```

As you can see, the main error is `rtabmap: Did not receive data since 5 seconds!`

I tried run nodes separately and played with different combinations for several hours. 

Any ideas what's wrong?

michelgokan, 2024-09-26:
Alright, I think I figured out the issue!

1. I need to copy the project into my own workspace (instead of relying on the pre-built package).
2. I need to change `/camera` to `/camera/camera` in the example files.
3. I have to rebuild the project and rerun it.
4. I found out that the [`realsense_d435i_color.launch.py`](https://raw.githubusercontent.com/introlab/rtabmap_ros/refs/heads/ros2/rtabmap_examples/launch/realsense_d435i_color.launch.py) file is too resource-heavy to run on a Raspberry Pi. It takes a few minutes before something appears on the screen. On the other hand, the [`realsense_d435i_stereo.launch.py`](https://raw.githubusercontent.com/introlab/rtabmap_ros/refs/heads/ros2/rtabmap_examples/launch/realsense_d435i_stereo.launch.py) file produces faster and better results (and without the 5-second warning).

I believe the 5-second warning occurs due to the overall computational intensity of RTAB-Map when processing color data.

closed 2024-09-26 by michelgokan

matlabbe, 2024-09-30:
You may not have needed to rebuild the full package. You can copy/paste the launch file from github or from the installed one on the computer, then modify the launch and start it directly with `ros2 launch my_modified_realsense_d435i_stereo.launch.py`.

Note sure why there is so much difference between color and stereo modes, they should run fairly the same, unless there is some synchronization issues or topic latency difference. Comparison on my desktop computer:
```
ros2 launch rtabmap_examples realsense_d435i_color.launch.py
[rgbd_odometry-2] [INFO] [1727657321.475066553] [rgbd_odometry]: Odom: quality=230, std dev=0.006382m|0.017944rad, update time=0.034914s delay=0.075090s
[rgbd_odometry-2] [INFO] [1727657321.544284506] [rgbd_odometry]: Odom: quality=239, std dev=0.006054m|0.021189rad, update time=0.034261s delay=0.077593s

ros2 launch rtabmap_examples realsense_d435i_stereo.launch.py
[stereo_odometry-2] [INFO] [1727657276.020022049] [stereo_odometry]: Odom: quality=257, std dev=0.019710m|0.022810rad, update time=0.049626s delay=0.083757s
[stereo_odometry-2] [INFO] [1727657276.085717890] [stereo_odometry]: Odom: quality=278, std dev=0.021567m|0.020040rad, update time=0.048890s delay=0.082494s
```
It is actually slower on stereo mode (because we need to compute the stereo correspondences).