introlab/rtabmap_ros #1215: A ROS2 example using the Robosense Helios 32 or Helios 16 (with rslidar_sdk)

mizrex, 2024-10-02:
I am trying to find a good example code using rtabmap with rslidar_sdk, for a Helios 32 (preferably). I am trying to build one using velodyne's example and some from ouster. But it doesn't seem to work. If anyone happens to have a working code, or a launch file. It'll be very helpful.

Thank you for all the help.

matlabbe, 2024-10-05:
Can you record a rosbag with the Helios 32 point cloud topics? Normally, it should be fairly similar to this [velodyne example](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/vlp16.launch.py). One thing different would be the actual point cloud topic, that I am not sure what it looks like with this lidar (a rosbag would help for that)

mizrex, 2024-10-08:
@matlabbe , thank you for getting back and advising. I was able to make it run using the Rtabmap launch file below. However, I noticed Rtabmap was relatively slow. I was hoping to try it later by reducing the number of channels from the Lidar. In the meantime, if you happen to notice any thing off in the code, please advise, or any suggestions. Thanks again for all the help you provide here!

`from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.conditions import IfCondition
from launch.substitutions import PathJoinSubstitution, LaunchConfiguration, TextSubstitution
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    return LaunchDescription([

        # Launch arguments
        DeclareLaunchArgument(
            name='use_sim_time',
            default_value='false',
            choices=['true','false'],
            description='Use simulation (Gazebo) clock if true'
        ),
        
        DeclareLaunchArgument(
            name='deskewing',
            default_value='false',
            choices=['true','false'],
            description='Enable lidar deskewing'
        ),

        DeclareLaunchArgument(
            name='localize_only',
            default_value='true',
            choices=['true','false'],
            description='Localize only, do not change loaded map'
        ),

        DeclareLaunchArgument(
            name='restart_map',
            default_value='false',
            choices=['true','false'],
            description='Delete previous map and restart'
        ),

        DeclareLaunchArgument(
            name='use_rtabmapviz',
            default_value='true',
            choices=['true','false'],
            description='Start rtabmapviz node'
        ),

        DeclareLaunchArgument(
            name='rtabmap_log_level',
            default_value='INFO',
            choices=['ERROR', 'WARN', 'INFO', 'DEBUG'],
            description='Set logger level for rtabmap.'
        ),

        DeclareLaunchArgument(
            name='icp_odometry_log_level',
            default_value='INFO',
            choices=['ERROR', 'WARN', 'INFO', 'DEBUG'],
            description='Set logger level for icp_odometry. Can set to WARN to reduce message output from this node.'
        ),

        # Nodes to launch
        Node(
            package='rtabmap_odom', executable='icp_odometry', output='screen',
            parameters=[{
                'frame_id':'rslidar',
                'odom_frame_id':'odom',
                'wait_for_transform':0.3, # 0.2
                'expected_update_rate':15.0,
[... 78 lines cut ...]
                'Icp/OutlierRatio', '0.7',
                'Icp/CorrespondenceRatio', '0.2',
                '--ros-args',
                '--log-level',
                [
                    TextSubstitution(text='rtabmap:='),
                    LaunchConfiguration('rtabmap_log_level'),
                ],
            ]), 
        Node(
            package='rtabmap_viz', executable='rtabmap_viz', output='screen',
            parameters=[{
                'frame_id':'rslidar',
                'odom_frame_id':'odom',
                'subscribe_odom_info':True,
                'subscribe_scan_cloud':True,
                'approx_sync':True, # False
                'use_sim_time':LaunchConfiguration('use_sim_time'),
            }],
            remappings=[
                ('scan_cloud', 'odom_filtered_input_scan')
            ],
            condition=IfCondition(LaunchConfiguration('use_rtabmapviz'))
        ),
    ])`

matlabbe, 2024-10-09:
> Rtabmap was relatively slow

Can you show the icp odometry log processing time here? In case if it is slow because of sync issue or that it is too much CPU for the machine. Setting `Icp/VoxelSize` and `OdomF2M/ScanSubtractRadius` higher (e..g, 0.3 - > 0.5) can help to be faster.

mizrex, 2024-10-29:
@matlabbe , your suggestion with the `VoxelSize `and `ScanSubtractRadius `worked like a charm!
Thanks again. Closing this.

closed 2024-10-29 by mizrex