introlab/rtabmap_ros #1217: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist.

smokemonkey101, 2024-10-07:
Im trying to run the example on the ros2 branch.
When i run rtabmap i get   "Could not transform IMU msg from frame "zed_imu_link" to frame "base_link", TF not available at time"
i dont understand how TF is not available 
Is my imu topic wrong?

When running the Zed wrapper with the ZED2i  i get a mapping failure
i need mapping for my purposes but it appears to start after trying a 2nd time.
`[zed.zed_node]: Mapping not activated: FAILURE`

there was a slight change to the launch file for the Zed wrapper compared to the RTABmap example "usage"
Im using 
> ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i

> ros2 launch rtabmap_launch rtabmap.launch.py \
>     rtabmap_args:="--delete_db_on_start" \
>     rgb_topic:=/zed/zed_node/rgb/image_rect_color \
>     depth_topic:=/zed/zed_node/depth/depth_registered \
>     camera_info_topic:=/zed/zed_node/rgb/camera_info \
>     frame_id:=base_link \
>     approx_sync:=false \
>     wait_imu_to_init:=true \
>     imu_topic:=/zed/zed_node/imu/data \
>     qos:=1 \
>     rviz:=true

Zed wrapper  readout with debug_mapping: true
```
jetson@jetson:~/ros2_ws$ ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i
[INFO] [launch]: All log files can be found below /home/jetson/.ros/log/2024-10-07-22-58-03-177256-jetson-112855
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [112874]
[INFO] [zed_wrapper-2]: process started with pid [112876]
[robot_state_publisher-1] [INFO] [1728341883.585956074] [zed.zed_state_publisher]: got segment zed_camera_center
[robot_state_publisher-1] [INFO] [1728341883.586132302] [zed.zed_state_publisher]: got segment zed_camera_link
[robot_state_publisher-1] [INFO] [1728341883.586156847] [zed.zed_state_publisher]: got segment zed_left_camera_frame
[robot_state_publisher-1] [INFO] [1728341883.586169359] [zed.zed_state_publisher]: got segment zed_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1728341883.586179791] [zed.zed_state_publisher]: got segment zed_right_camera_frame
[robot_state_publisher-1] [INFO] [1728341883.586188848] [zed.zed_state_publisher]: got segment zed_right_camera_optical_frame
[zed_wrapper-2] [INFO] [1728341883.654548566] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1728341883.654809212] [zed.zed_node]:       ZED Camera Component 
[zed_wrapper-2] [INFO] [1728341883.654834653] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1728341883.654849469] [zed.zed_node]:  * namespace: /zed
[zed_wrapper-2] [INFO] [1728341883.654861310] [zed.zed_node]:  * node name: zed_node
[zed_wrapper-2] [INFO] [1728341883.654871486] [zed.zed_node]: ********************************
[zed_wrapper-2] [INFO] [1728341883.705131186] [zed.zed_node]: *** DEBUG parameters ***
[zed_wrapper-2] [INFO] [1728341883.705346072] [zed.zed_node]:  * SDK Verbose: 1
[zed_wrapper-2] [INFO] [1728341883.705415513] [zed.zed_node]:  * Debug Common: FALSE
[zed_wrapper-2] [INFO] [1728341883.705448250] [zed.zed_node]:  * Debug Simulation: FALSE
[zed_wrapper-2] [INFO] [1728341883.705473211] [zed.zed_node]:  * Debug Video/Depth: FALSE
[zed_wrapper-2] [INFO] [1728341883.705499963] [zed.zed_node]:  * Debug Control settings: FALSE
[zed_wrapper-2] [INFO] [1728341883.705521948] [zed.zed_node]:  * Debug Point Cloud: FALSE
[zed_wrapper-2] [INFO] [1728341883.705540796] [zed.zed_node]:  * Debug GNSS: FALSE
[zed_wrapper-2] [INFO] [1728341883.705559133] [zed.zed_node]:  * Debug Positional Tracking: FALSE
[zed_wrapper-2] [INFO] [1728341883.705582045] [zed.zed_node]:  * Debug sensors: FALSE
[zed_wrapper-2] [INFO] [1728341883.705600574] [zed.zed_node]:  * Debug Mapping: TRUE
[zed_wrapper-2] [INFO] [1728341883.705616862] [zed.zed_node]:  * Debug Object Detection: FALSE
[zed_wrapper-2] [INFO] [1728341883.705633086] [zed.zed_node]:  * Debug Body Tracking: FALSE
[zed_wrapper-2] [INFO] [1728341883.705652735] [zed.zed_node]:  * Debug Streaming: FALSE
[zed_wrapper-2] [INFO] [1728341883.705670591] [zed.zed_node]:  * Debug ROI: FALSE
[zed_wrapper-2] [INFO] [1728341883.705687680] [zed.zed_node]:  * Debug Advanced: FALSE
[zed_wrapper-2] [INFO] [1728341883.705714208] [zed.zed_node]:  + Debug Mode enabled +
[zed_wrapper-2] [INFO] [1728341883.705756897] [zed.zed_node]: *** GENERAL parameters ***
[zed_wrapper-2] [INFO] [1728341883.705827875] [zed.zed_node]:  * Camera model: zed2i - ZED 2i
[zed_wrapper-2] [INFO] [1728341883.705868196] [zed.zed_node]:  * Camera name: zed
[zed_wrapper-2] [INFO] [1728341883.705894277] [zed.zed_node]:  * Camera SN: 0
[zed_wrapper-2] [INFO] [1728341883.705921989] [zed.zed_node]:  * Camera timeout [sec]: 5
[zed_wrapper-2] [INFO] [1728341883.705945638] [zed.zed_node]:  * Camera reconnection temptatives: 5
[zed_wrapper-2] [INFO] [1728341883.705967238] [zed.zed_node]:  * Camera framerate: 30
[zed_wrapper-2] [INFO] [1728341883.705986311] [zed.zed_node]:  * GPU ID: -1
[zed_wrapper-2] [INFO] [1728341883.706032200] [zed.zed_node]:  * Camera resolution: HD1080
[... 453 lines cut ...]
[rtabmap_viz-3] [INFO] [1728340884.091191982] [rtabmap.rtabmap_viz]: rtabmap_viz: subscribe_user_data = false
[rtabmap_viz-3] [INFO] [1728340884.091197838] [rtabmap.rtabmap_viz]: rtabmap_viz: subscribe_scan = false
[rtabmap_viz-3] [INFO] [1728340884.091203887] [rtabmap.rtabmap_viz]: rtabmap_viz: subscribe_scan_cloud = false
[rtabmap_viz-3] [INFO] [1728340884.091209711] [rtabmap.rtabmap_viz]: rtabmap_viz: subscribe_scan_descriptor = false
[rtabmap_viz-3] [INFO] [1728340884.091215503] [rtabmap.rtabmap_viz]: rtabmap_viz: topic_queue_size = 1
[rtabmap_viz-3] [INFO] [1728340884.091221327] [rtabmap.rtabmap_viz]: rtabmap_viz: sync_queue_size  = 10
[rtabmap_viz-3] [INFO] [1728340884.091227344] [rtabmap.rtabmap_viz]: rtabmap_viz: qos_image       = 1
[rtabmap_viz-3] [INFO] [1728340884.091233392] [rtabmap.rtabmap_viz]: rtabmap_viz: qos_camera_info = 1
[rtabmap_viz-3] [INFO] [1728340884.091239152] [rtabmap.rtabmap_viz]: rtabmap_viz: qos_scan        = 1
[rtabmap_viz-3] [INFO] [1728340884.091244688] [rtabmap.rtabmap_viz]: rtabmap_viz: qos_odom        = 1
[rtabmap_viz-3] [INFO] [1728340884.091250416] [rtabmap.rtabmap_viz]: rtabmap_viz: qos_user_data   = 1
[rtabmap_viz-3] [INFO] [1728340884.091256081] [rtabmap.rtabmap_viz]: rtabmap_viz: approx_sync     = false
[rtabmap_viz-3] [INFO] [1728340884.091272401] [rtabmap.rtabmap_viz]: Setup depth callback
[rgbd_odometry-1] [ERROR] [1728340884.102131105] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "base_link", TF not available at time 1728340883.630996
[rtabmap_viz-3] [INFO] [1728340884.107204060] [rtabmap.rtabmap_viz]: 
[rtabmap_viz-3] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-3]    /rtabmap/odom \
[rtabmap_viz-3]    /zed/zed_node/rgb/image_rect_color \
[rtabmap_viz-3]    /zed/zed_node/depth/depth_registered \
[rtabmap_viz-3]    /zed/zed_node/rgb/camera_info \
[rtabmap_viz-3]    /rtabmap/odom_info
[rtabmap_viz-3] [INFO] [1728340884.112100977] [rtabmap.rtabmap_viz]: rtabmap_viz started.
[rgbd_odometry-1] [ERROR] [1728340884.305210301] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed_imu_link" to frame "base_link", TF not available at time 1728340883.843416
[rgbd_odometry-1] [ WARN] (2024-10-07 22:41:21.819) MsgConversion.cpp:1994::getTransform() (can transform base_link -> zed_imu_link?) Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist. canTransform returned after 0.203065 timeout was 0.2. (wait_for_transform=0.200000)
```

matlabbe, 2024-10-08:
Try this (more recent) zed example: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/zed.launch.py

I think zed changed slightly their launch file since the example on the main README.

smokemonkey101, 2024-10-08:
Ah that does run.
For some reason the zed.launch.py file wasn't included when i installed using the apt install method. I just manually download it and added it. 

**Now the problem im getting is the camera flickers strangely. Even the depth map.**

I i tried with use zed odom true and false and with approx_sync True and False
'approx_sync':False,
'use_zed_odometry', default_value='true'
```
[rtabmap-4] [INFO] [1728406742.833590748] [rtabmap]: rtabmap (101): Rate=1.00s, Limit=0.000s, Conversion=0.0019s, RTAB-Map=0.0916s, Maps update=0.0001s pub=0.0016s (local map=4, WM=4)
[rtabmap-4] [WARN] [1728406743.107728087] [rtabmap]: We received odometry message, but we cannot get the corresponding TF odom->zed_camera_link at data stamp 1728406742.837028s (odom msg stamp is 1728406742.837028s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once.
```
```
[WARN] [1728407211.838314867] [rgbd_sync]: The time difference between rgb and depth frames is high (diff=0.030159s, rgb=1728407211.769395s, depth=1728407211.799554s). You may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.
```
[Screencast Flicker.webm](https://github.com/user-attachments/assets/0d282ac1-758f-494a-b29c-cc9843ac3cf6)

smokemonkey101, 2024-10-08:
Sorry it turns out it was mt PC weird. it works just fin on my Jetson Orin

matlabbe, 2024-10-09:
Those image corruptions are pretty bad, hopefully it works on your orin.