introlab/rtabmap_ros #1218: In rgbd_odometry.cpp Error: no matching function for call to ‘message_filters::Subscriber<rtabmap_msgs::msg::RGBDImage_<std::allocator<void> > >::subscribe(rtabmap_odom::RGBDOdometry*, const char [12], rmw_qos_profile_t&, rclcpp::SubscriptionOptions&)’   137 | QoS(topicQueueSize_).reliability((rmw_qos_reliability_policy_t)qos()).get_rmw_qos_profile(), options);

7yzx, 2024-10-09:
my device is jetson agx xavier, ubuntu20,Jetpack5 use ros2 foxy

I can't find a solution to these error. Need Help! 

```/home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp: In member function ‘virtual void rtabmap_odom::RGBDOdometry::onOdomInit()’:
/home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:137:160: error: no matching function for call to ‘message_filters::Subscriber<rtabmap_msgs::msg::RGBDImage_<std::allocator<void> > >::subscribe(rtabmap_odom::RGBDOdometry*, const char [12], rmw_qos_profile_t&, rclcpp::SubscriptionOptions&)’
  137 | QoS(topicQueueSize_).reliability((rmw_qos_reliability_policy_t)qos()).get_rmw_qos_profile(), options);
      |                                                                                                     ^

In file included from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/include/rtabmap_odom/rgbd_odometry.hpp:33,
                 from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:28:
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node::SharedPtr, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t]’
  145 |   void subscribe(rclcpp::Node::SharedPtr node, const std::string& topic, const rmw_qos_profile_t qos = rmw_qos_profile_default)
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note:   candidate expects 3 arguments, 4 provided
/opt/ros/foxy/include/message_filters/subscriber.h:162:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node*, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t]’
  162 |   void subscribe(rclcpp::Node * node, const std::string& topic, const rmw_qos_profile_t qos = rmw_qos_profile_default)
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:162:8: note:   candidate expects 3 arguments, 4 provided
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe() [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >]’
  184 |   void subscribe()
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note:   candidate expects 0 arguments, 4 provided
/home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:138:160: error: no matching function for call to ‘message_filters::Subscriber<rtabmap_msgs::msg::RGBDImage_<std::allocator<void> > >::subscribe(rtabmap_odom::RGBDOdometry*, const char [12], rmw_qos_profile_t&, rclcpp::SubscriptionOptions&)’
  138 | QoS(topicQueueSize_).reliability((rmw_qos_reliability_policy_t)qos()).get_rmw_qos_profile(), options);
      |                                                                                                     ^

In file included from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/include/rtabmap_odom/rgbd_odometry.hpp:33,
                 from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:28:
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node::SharedPtr, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t]’
  145 |   void subscribe(rclcpp::Node::SharedPtr node, const std::string& topic, const rmw_qos_profile_t qos = rmw_qos_profile_default)
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note:   candidate expects 3 arguments, 4 provided
/opt/ros/foxy/include/message_filters/subscriber.h:162:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node*, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t]’
  162 |   void subscribe(rclcpp::Node * node, const std::string& topic, const rmw_qos_profile_t qos = rmw_qos_profile_default)
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:162:8: note:   candidate expects 3 arguments, 4 provided
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe() [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >]’
  184 |   void subscribe()
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note:   candidate expects 0 arguments, 4 provided
/home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:141:161: error: no matching function for call to ‘message_filters::Subscriber<rtabmap_msgs::msg::RGBDImage_<std::allocator<void> > >::subscribe(rtabmap_odom::RGBDOdometry*, const char [12], rmw_qos_profile_t&, rclcpp::SubscriptionOptions&)’
  141 | QoS(topicQueueSize_).reliability((rmw_qos_reliability_policy_t)qos()).get_rmw_qos_profile(), options);
      |                                                                                                     ^

In file included from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/include/rtabmap_odom/rgbd_odometry.hpp:33,
                 from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:28:
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node::SharedPtr, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t]’
  145 |   void subscribe(rclcpp::Node::SharedPtr node, const std::string& topic, const rmw_qos_profile_t qos = rmw_qos_profile_default)
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note:   candidate expects 3 arguments, 4 provided
/opt/ros/foxy/include/message_filters/subscriber.h:162:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node*, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t]’
  162 |   void subscribe(rclcpp::Node * node, const std::string& topic, const rmw_qos_profile_t qos = rmw_qos_profile_default)
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:162:8: note:   candidate expects 3 arguments, 4 provided
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe() [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >]’
  184 |   void subscribe()
      |        ^~~~~~~~~
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note:   candidate expects 0 arguments, 4 provided
/home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:145:161: error: no matching function for call to ‘message_filters::Subscriber<rtabmap_msgs::msg::RGBDImage_<std::allocator<void> > >::subscribe(rtabmap_odom::RGBDOdometry*, const char [12], rmw_qos_profile_t&, rclcpp::SubscriptionOptions&)’
  145 | QoS(topicQueueSize_).reliability((rmw_qos_reliability_policy_t)qos()).get_rmw_qos_profile(), options);
      |                                                                                                     ^

In file included from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/include/rtabmap_odom/rgbd_odometry.hpp:33,
                 from /home/zen/ROS2_SLAM_WS/src/rtabmap_ros/rtabmap_odom/src/nodelets/rgbd_odometry.cpp:28:
/opt/ros/foxy/include/message_filters/subscriber.h:145:8: note: candidate: ‘void message_filters::Subscriber<M>::subscribe(rclcpp::Node::SharedPtr, const string&, rmw_qos_profile_t) [with M = rtabmap_msgs::msg::RGBDImage_<std::allocator<void> >; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>; std::string = std::__cxx11::basic_string<char> [... cut ...]
/opt/ros/foxy/include/message_filters/subscriber.h:184:8: note:   candidate expects 0 arguments, 4 provided
make[2]: *** [CMakeFiles/rtabmap_odom_plugins.dir/build.make:63: CMakeFiles/rtabmap_odom_plugins.dir/src/nodelets/rgbd_odometry.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:113: CMakeFiles/rtabmap_odom_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rtabmap_odom [2min 7s, exited with code 2]

Summary: 14 packages finished [2min 47s]
  1 package failed: rtabmap_odom
  1 package had stderr output: rtabmap_odom
  4 packages not processed

```

matlabbe, 2024-10-10:
Either use [foxy-devel branch](https://github.com/introlab/rtabmap_ros/tree/foxy-devel), or try this [version](https://github.com/introlab/rtabmap_ros/tree/268609b0b70b97bb56c30985d397345c0dea753b) just before we added `rclcpp::SubscriptionOptions` and callback group in this [one](https://github.com/introlab/rtabmap_ros/commit/c956e3780fa2e2e1639ab583b7fe5dcaec644c50) (which seems related to your error).

7yzx, 2024-10-10:
thank you, all right now!