introlab/rtabmap_ros #1223: How to call up a 3D map in rviz?

miku54, 2024-10-16:
Hello everyone!
--I am debugging the RTABMAP Slam function on Jetson Orin nano. I use Astra camera instead of Intel D435*. Although the camera has no odometry data, I publish the odometry topic named odom_combined from another node.
--My RGB camera (/camera/color/image_raw) and depth camera (/camera/depth/image_raw) data size are both 640×480.
Refer to https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_demos/launch/turtlebot3_rgbd_sync.launch.py ​​file to modify the camera data. After running rtabmap, open rviz and no 3D MAP appears.
My file:

```
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable
from launch.substitutions import LaunchConfiguration
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node

def generate_launch_description():

    use_sim_time = LaunchConfiguration('use_sim_time')
    qos = LaunchConfiguration('qos')
    localization = LaunchConfiguration('localization')

    icp_parameters={
          'odom_frame_id':'odom_combined',
          'qos':qos
    }

    rtabmap_parameters={
          'subscribe_rgbd':True,
          'subscribe_scan':True,
          'use_action_for_goal':True,
          'odom_sensor_sync': True,
          'qos_scan':qos,
          'qos_image':qos,
          'qos_imu':qos,
          # RTAB-Map's parameters should be strings:
          'Mem/NotLinkedNodesKept':'false'
    }

    # Shared parameters between different nodes
    shared_parameters={
          'frame_id':'camera_link',
          'use_sim_time':use_sim_time,
          # RTAB-Map's parameters should be strings:
          'Reg/Strategy':'1',
          'Reg/Force3DoF':'true',
          'Mem/NotLinkedNodesKept':'false',
          'Icp/PointToPlaneMinComplexity':'0.04' # to be more robust to long corridors with low geometry
    }

    remappings=[
          ('odom', '/odom_combined'),
          ('scan', '/scan'),
          ('rgb/image', '/camera/color/image_raw'), 
          ('rgb/camera_info', '/camera/color/camera_info'),
          ('depth/image', '/camera/depth/image_raw')]

    return LaunchDescription([

        # Launch arguments
        DeclareLaunchArgument(
            'use_sim_time', default_value='false', choices=['true', 'false'],
            description='Use simulation (Gazebo) clock if true'),
        
        DeclareLaunchArgument(
            'qos', default_value='0',
            description='QoS used for input sensor topics'),
            
        DeclareLaunchArgument(
            'localization', default_value='false', choices=['true', 'false'],
            description='Launch rtabmap in localization mode (a map should have been already created).'),

        # Nodes to launch
        Node(
            package='rtabmap_sync', executable='rgbd_sync', output='screen',
            parameters=[{'approx_sync':False, 'use_sim_time':use_sim_time, 'qos':qos}],
            remappings=remappings),

        Node(
            package='rtabmap_odom', executable='icp_odometry', output='screen',
            parameters=[icp_parameters, shared_parameters],
            remappings=remappings,
            arguments=["--ros-args", "--log-level", 'icp_odometry:=warn']),

        # SLAM Mode:
        Node(
            condition=UnlessCondition(localization),
            package='rtabmap_slam', executable='rtabmap', output='screen',
            parameters=[rtabmap_parameters, shared_parameters],
            remappings=remappings,
            arguments=['-d']),
            
        # Localization mode:
        Node(
            condition=IfCondition(localization),
            package='rtabmap_slam', executable='rtabmap', output='screen',
            parameters=[rtabmap_parameters, shared_parameters,
              {'Mem/IncrementalMemory':'False',
               'Mem/InitWMWithAllNodes':'True'}],
            remappings=remappings),

    ])
```

--Use the following command to run rtabmap with the same effect:
 ros2 launch rtabmap_launch rtabmap.launch.py ​​visual_odometry:=false \ rtabmap_args:="--delete_db_on_start" \ rgb_topic:=/camera/color/image_raw \ depth_topic:=/camera/depth/image_raw \ camera_info_topic :=/camera/color/camera_info \ odom_topic:=/odom_combined \ frame_id:=base_footprint \ approx_sync:=true \ rtabmap_viz:=false \ wait_imu_to_init:=false \ qos:=1 \ rviz:=false

Expected Results：
[image]
Actual Results：
[image]

matlabbe, 2024-10-17:
Try adding `rtabmap_ros/MapCloud` display. It looks what it was used in the other screen shot to show the 3D point cloud.

miku54, 2024-10-17:
> Try adding `rtabmap_ros/MapCloud` display. It looks what it was used in the other screen shot to show the 3D point cloud.

I tried adding "MapCloud" from "add->by display type->rtabmap_rviz_plugins", and selecting /rtabmap/mapData or /mapData, but it didn't show up. Is my approach wrong or is there something wrong? Please help me, I can't find a solution.
[image]

matlabbe, 2024-10-17:
Can you record a rosbag with `/rtabmap/mapData`? and share the corresponding `~/.ros/rtabmap.db` created?

miku54, 2024-10-17:
> Can you record a rosbag with `/rtabmap/mapData`? and share the corresponding `~/.ros/rtabmap.db` created?

Thanks for your reply. This is a piece of data I recorded. If you need more information, please contact me.
https://drive.google.com/file/d/11ot1RBlt3gAa9zuNDejJr3MTBuwsawN9/view?usp=drive_link

miku54, 2024-10-18:
> raw images

Thank you for your analysis. I did not do camera calibration before running rtab. Is it necessary to do camera calibration? Please wait for the raw images data after moving. I will upload it in 2 hours. Thank you again for your reply.

miku54, 2024-10-18:
> There is something wrong with the calibration, rtabmap doesn't see it as valid for some reason, so clouds cannot be created from depth (note also that depth is not always valid): ![2024-10-17_20-37](https://private-user-images.githubusercontent.com/2319645/377711956-d94b6303-9113-4ebe-8db2-79d175d5d7d8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkyMjMzNDMsIm5iZiI6MTcyOTIyMzA0MywicGF0aCI6Ii8yMzE5NjQ1LzM3NzcxMTk1Ni1kOTRiNjMwMy05MTEzLTRlYmUtOGRiMi03OWQxNzVkNWQ3ZDgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMThUMDM0NDAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGZhNGQyZjExY2Y0MDQ0MmVlZGI0NzA5ZDZkMDZlMDY2N2FlMzcxOWQzYmYwMTIwYzIzYWI0YmRhMjU1ZGViMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.awUycw2KHplUFSIGn8xQDNu_TUWtlAFLuEeqG-oGEQ0)
> 
> Can you record raw images while moving slightly? `ros2 bag record /camera/color/camera_info /camera/color/image_raw /camera/depth/image_raw`
Thank you for your analysis. I did not do camera calibration before running rtab. Is it necessary to do camera calibration? Please wait for the raw images data after moving. I will upload it in 2 hours. Thank you again for your reply.

matlabbe, 2024-10-18:
You may not need to calibrate, if the camera driver already provides one. I would expect that the Astra RGB-D camera provides already that kind of information, but that's why I want to look in the camera_info. The depth images look strange too. Maybe there is an issue with the camera/driver.

miku54, 2024-10-18:
> You may not need to calibrate, if the camera driver already provides one. I would expect that the Astra RGB-D camera provides already that kind of information, but that's why I want to look in the camera_info. The depth images look strange too. Maybe there is an issue with the camera/driver.

Sorry for the delay due to some personal matters. Here are the data I recorded and the corresponding db map file. If you need other data, I can provide it at any time:
https://drive.google.com/file/d/1rMxsKN1mARFVBBW6QMp_NRpZnHZKizUP/view?usp=sharing

miku54, 2024-10-18:
> > You may not need to calibrate, if the camera driver already provides one. I would expect that the Astra RGB-D camera provides already that kind of information, but that's why I want to look in the camera_info. The depth images look strange too. Maybe there is an issue with the camera/driver.
> 
> Sorry for the delay due to some personal matters. Here are the data I recorded and the corresponding db map file. If you need other data, I can provide it at any time: https://drive.google.com/file/d/1rMxsKN1mARFVBBW6QMp_NRpZnHZKizUP/view?usp=sharing
> 
> I observed that the error "util3d.cpp:1186::cloudsRGBFromSensorData() Camera model 0 is invalid" would appear during the run. I referred to #866, but it did not help me. ![image](https://private-user-images.githubusercontent.com/81612531/377801425-2a05688e-211e-4a92-8fd7-ad7a5b0c150b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkyNDM2MzQsIm5iZiI6MTcyOTI0MzMzNCwicGF0aCI6Ii84MTYxMjUzMS8zNzc4MDE0MjUtMmEwNTY4OGUtMjExZS00YTkyLThmZDctYWQ3YTViMGMxNTBiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE4VDA5MjIxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5YjZjMzEyYTFiZTQxOTllMWUzYmE0MmYyMTQ5NzlmODEzZmE1MWQ3MmE5NjFhMDIwNTQ4MzVkM2Y5OWYyOWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.obEZyV3w2OhUqvmQArmB6e-oV53GiXEwU9YT3GzKZDI)

Added:
I have saved other db map files and used rtabmap-database tool to check [Calib], all of them are NA, so I think there is no problem with my camera driver.

matlabbe, 2024-10-19:
The camera_info is wrong, this is an invalid projection matrix
```
p:
- 570.3422047415297
- 0.0
- 0.0
- 0.0
- 0.0
- 570.3422047415297
- 570.3422047415297
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
```
It should looks like this (based on k):
```
p:
- 570.3422047415297
- 0.0
- 319.5
- 0.0
- 0.0
- 570.3422047415297
- 239.5
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
```

miku54, 2024-10-22:
> The camera_info is wrong, this is an invalid projection matrix
> 
> ```
> p:
> - 570.3422047415297
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 570.3422047415297
> - 570.3422047415297
> - 0.0
> - 0.0
> - 0.0
> - 1.0
> - 0.0
> ```
> 
> It should look like this (based on k):
> 
> ```
> p:
> - 570.3422047415297
> - 0.0
> - 319.5
> - 0.0
> - 0.0
> - 570.3422047415297
> - 239.5
> - 0.0
> - 0.0
> - 0.0
> - 1.0
> - 0.0
> ```

Thanks for your reply. When I tested with other cameras, the camere_info data looked normal. After running rtab map, I could see the 3D map in rviz2. Thanks again for your help.😊

closed 2024-10-22 by miku54

kowyo, 2024-11-01:
Hi @matlabbe and @miku54, good day. I have read the above disscusion as well as #866. I am using the exactly same platform as @miku54. From my point of view this seems to be a issue related to the topic of `/camera/color/camera_info`, instead of the type of camera, so this issue isn't really fixed.

I am wondering that, after knowing it is related to projection matrix, is there any direction I can start working on? Like change the source code of camera file, or ask the vendor for the correct driver or detailed help.

Which is a correct way I should go?

Thank you!

matlabbe, 2024-11-01:
Is the camera driver open source (at least the ROS part)? Can you link it here? 

You may look at how the camera_info is published by the camera driver, then figure why the projection matrix is wrong.

miku54, 2024-11-05:
> Hi @matlabbe and @miku54, good day. I have read the above disscusion as well as #866. I am using the exactly same platform as @miku54. From my point of view this seems to be a issue related to the [...]

It is indeed a problem with /camera/color/camera_info. After I changed the camera driver, the /camera/color/camera_info data became normal, and the 3D point cloud can be seen normally when running rtabmap. I also suggest you check whether your /camera/color/camera_info data is normal. If it is not normal,maybe  you can manually subscribe to this topic and modify the data and publish it again to make it work normally.Good luck with your debugging!(*^_^*)

kowyo, 2024-11-07:
Hi, community users and maintainers. I solved `camera model 0 is invalid`.

At first, I launched rtabmap_viz with command `ros2 run rtabmap_viz rtabmap_viz` in a seperated terminal, which did not have the remappings of camera info and therefore caused the error.

A better implementation is to laucn rtabmap_viz in the launch file where you launch rtabmap as following (in <rtabmap_launch_file_name>.launch.py):

```python
Node(
    package='rtabmap_viz', executable='rtabmap_viz', output='screen',
    remappings=remappings),
```

Where `remappings` contain your own remappings of the topics.

kowyo, 2024-11-07:
Still, 3D map is still missing, which is due to incorrect camera_info given. To be specific, it is due to an error projection matrix given. Ask for a correct driver from vendor may be the time-saving approach for you.