introlab/rtabmap_ros #1226: [ERROR] [stereo_odometry-2]: process has died

pangpangxiong520, 2024-10-22:
##My Device
Realsense D435i
ROS2 humble
ubuntu 22.04 LTS
When I clone the codes in ros2 branch, I just follow the readme.md and run the command ros2 launch rtabmap_examples realsense_d435i_color.launch.py
The realsense node is successfully up:
[image]
However, there is an error and stereo_odometry process has died:
[image]
Then the gui comes up to the screen but here comes the warning:
[image]
And when I try to see the topic /odom with command ros2 topic hz /odom, there is nothing appear.
However, the topic  /imu/imu_data can output information:
[image]
So what's the problem?:(

pangpangxiong520, 2024-10-23:
I resolved this error by reinstalling RTABMAP, but I can't rule out the possibility that installing the turtlebot3 package before reinstalling RTABMAP also contributed to the solution. Perhaps due to dependencies on turtlebot3, the problem was fixed after reinstalling RTABMAP. I'll go ahead and close this issue.

closed 2024-10-23 by pangpangxiong520

matlabbe, 2024-10-26:
Looks like you have a custom opencv version, this can cause problem with rtabmap_ros if those opencv libraries are loaded on runtime and they are not compatible with the ones rtabmap is built on.