introlab/rtabmap_ros #1227: Did not receive data since 5 seconds!

ofarukdemirtas, 2024-10-25:
I am trying to run rtabmap with zed2i camera and livox lidar but i got this error 
[ WARN] [1729862098.568035980]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=50 and 10 respectively).
/rtabmap/rtabmap_viz subscribed to:
   /zed/zed_node/odom
[ WARN] [1729862102.520261674]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap/rtabmap subscribed to (approx sync):
   /zed/zed_node/odom \
   /zed/zed_node/rgb/image_rect_color \
   /zed/zed_node/depth/depth_registered \
   /zed/zed_node/rgb/camera_info \
   /rtabmap/scan
[ WARN] [1729862104.568016242]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=50 and 10 respectively).
/rtabmap/rtabmap_viz subscribed to:
   /zed/zed_node/odom
[ WARN] [1729862108.520232466]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap/rtabmap subscribed to (approx sync):
   /zed/zed_node/odom \
   /zed/zed_node/rgb/image_rect_color \
   /zed/zed_node/depth/depth_registered \
   /zed/zed_node/rgb/camera_info \
   /rtabmap/scan
[ WARN] [1729862110.568043996]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=50 and 10 respectively).
/rtabmap/rtabmap_viz subscribed to:
   /zed/zed_node/odom
[ WARN] [1729862114.520246735]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap/rtabmap subscribed to (approx sync):
   /zed/zed_node/odom \
   /zed/zed_node/rgb/image_rect_color \
   /zed/zed_node/depth/depth_registered \
   /zed/zed_node/rgb/camera_info \
   /rtabmap/scan
[ WARN] [1729862116.568090770]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=50 and 10 respectively).
/rtabmap/rtabmap_viz subscribed to:
   /zed/zed_node/odom

I dont understand and dont know how to solve this is also my rostopic and node liist 

tiago@tmsp:~/catkin_ws$ rosnode list
/livox_lidar_publisher2
/livox_tf_broadcaster
/rosout
/rtabmap/rtabmap
/rtabmap/rtabmap_viz
/rviz
/zed/zed_node
/zed/zed_state_publisher
tiago@tmsp:~/catkin_ws$ rostopic list
/clicked_point
/diagnostics
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/rtabmap/cloud_ground
/rtabmap/cloud_map
/rtabmap/cloud_obstacles
/rtabmap/elevation_map
/rtabmap/global_path
/rtabmap/global_path_nodes
/rtabmap/global_pose
/rtabmap/goal
/rtabmap/goal_node
/rtabmap/goal_out
/rtabmap/goal_reached
/rtabmap/gps/fix
/rtabmap/grid_map
/rtabmap/grid_prob_map
/rtabmap/imu
/rtabmap/info
/rtabmap/initialpose
[... 288 lines cut ...]
            <param name="RGBD/ProximityPathMaxNeighbors" type="int" value="1" />
            <param name="Grid/RangeMax" type="double" value="15.0" />
            <param name="Vis/MinInliers" type="int" value="10" />
        </node>

        
        <node name="rtabmap_viz" pkg="rtabmap_viz" type="rtabmap_viz" output="screen">
            <remap from="rgb/image" to="/$(arg camera_name)/$(arg zed_node_name)/rgb/image_rect_color" />
            <remap from="depth/image" to="/$(arg camera_name)/$(arg zed_node_name)/depth/depth_registered" />
            <remap from="rgb/camera_info" to="/$(arg camera_name)/$(arg zed_node_name)/rgb/camera_info" />
            <remap from="cloud" to="/livox/lidar" />
            <remap from="odom" to="/$(arg camera_name)/$(arg zed_node_name)/odom" />
            
            
            <param name="queue_size" type="int" value="50" />
            <param name="approx_sync" type="bool" value="true" />
            <param name="approx_sync_max_interval" type="double" value="0.5" />
        </node>
    </group>

    
    <group if="$(arg rviz_enable)">
        <node name="rviz" pkg="rviz" type="rviz" respawn="true" output="screen"
              args="-d $(find ugv_navigation)/rviz/navigation.rviz" />
    </group>`

ofarukdemirtas, 2024-10-25:
`[ WARN] [1729869211.466197990]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap/rtabmap subscribed to (approx sync):
   /zed/zed_node/odom \
   /zed/zed_node/rgb/image_rect_color \
   /zed/zed_node/depth/depth_registered \
   /zed/zed_node/rgb/camera_info \
   /rtabmap/scan
[2024-10-25 16:13:31.592] [console] [info]  Receive Command: Id 258 Seq 228  [hap_command_handler.cpp] [Handle] [67]
[2024-10-25 16:13:31.729] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD227EA1002W1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[2024-10-25 16:13:32.592] [console] [info]  Receive Command: Id 258 Seq 229  [hap_command_handler.cpp] [Handle] [67]
[2024-10-25 16:13:32.729] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD227EA1002W1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[ WARN] [1729869213.505161120]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=50 and 10 respectively).
/rtabmap/rtabmap_viz subscribed to:
   /zed/zed_node/odom
[2024-10-25 16:13:33.592] [console] [info]  Receive Command: Id 258 Seq 230  [hap_command_handler.cpp] [Handle] [67]
[2024-10-25 16:13:33.728] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD227EA1002W1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[2024-10-25 16:13:34.592] [console] [info]  Receive Command: Id 258 Seq 231  [hap_command_handler.cpp] [Handle] [67]
[2024-10-25 16:13:34.729] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD227EA1002W1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[2024-10-25 16:13:35.592] [console] [info]  Receive Command: Id 258 Seq 232  [hap_command_handler.cpp] [Handle] [67]
[2024-10-25 16:13:35.729] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD227EA1002W1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[2024-10-25 16:13:36.592] [console] [info]  Receive Command: Id 258 Seq 233  [hap_command_handler.cpp] [Handle] [67]
[2024-10-25 16:13:36.729] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD227EA1002W1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[ WARN] [1729869217.466196753]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap/rtabmap subscribed to (approx sync):
   /zed/zed_node/odom \
   /zed/zed_node/rgb/image_rect_color \
   /zed/zed_node/depth/depth_registered \
   /zed/zed_node/rgb/camera_info \
   /rtabmap/scan
` I took hap lidar data perfectly but i cannot take the data from rtabmap i dont understand the issue

matlabbe, 2024-10-26:
What is the topic name of the lidar point cloud topic? For rtabmap node, just set `subscribe_scan_cloud` to true (remove `subscribe_scan`), then you will need to remap scan_cloud to the lidar point cloud topic:
```xml
<node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen">
    ...
    <remap from="scan_cloud" to="{hap_point_cloud_topic}"/>
```

I also don't recommend to sync lidar and all camera topics like this to rtabmap node. Use `rgbd_sync` node and use `subscribe_rgbd_image` for rtabmap node. See example [here](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_2D_laser).

Also for the zed config, set parameter to disable map TF publishing.