introlab/rtabmap_ros #1231: Fatal Error on Database in Multi-Session Mapping with LIDAR and AprilTags

brayanpa, 2024-10-29:
Hi Mathieu,

We’re experiencing some issues with mapping using a single LIDAR configuration along with AprilTags.

In the first mapping session, everything works smoothly, regardless of the number of AprilTags detected. However, when a new session starts, we can switch to localization mode without any issues. The problem arises when we attempt to close the database—it throws an error, and the database then becomes corrupted.

In some cases, the error occurs when closing the database, but the database continues to function without noticeable issues. Below, I’ve included some logs we've captured, as well as a sample of one of the corrupted databases.

Please let us know if there’s any additional information we can provide to help in troubleshooting this bug.

We are currently working with these versions:

rtabmap: [f3ccfcb45219fbbb40648072a8b9adb0f6143e77](https://github.com/introlab/rtabmap/commit/f3ccfcb45219fbbb40648072a8b9adb0f6143e77)
rtabmap-ros: 28f997477aa80130da4c0915d4f0182280c1a9a7

Thank you!

[rtabmap_log1.txt](https://github.com/user-attachments/files/17558248/rtabmap_log1.txt)
[rtabmap_log2.txt](https://github.com/user-attachments/files/17558250/rtabmap_log2.txt)
[map.original.db.zip](https://github.com/user-attachments/files/17558268/map.original.db.zip)
[rtabmap_params.txt](https://github.com/user-attachments/files/17558465/rtabmap_params.txt)

brayanpa, 2024-10-29:
We have discovered some additional information. Occasionally, when mapping a new session and switching to localization mode (to finish the map), the database file disappears. Typically, it only disappears momentarily, but in some cases, it completely vanishes. Here are the logs captured during these instances:

```[rtabmap-1] [INFO] [1730234043.773160352] [gary.nav.rtabmap.rtabmap]: rtabmap (303): Rate=1.00s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.0045s, Maps update=0.0020s pub=0.0003s delay=0.1209s (local map=163, WM=160)
[rtabmap-1] [INFO] [1730234044.822210010] [gary.nav.rtabmap.rtabmap]: rtabmap (304): Rate=1.00s, Limit=0.000s, Conversion=0.0012s, RTAB-Map=0.0049s, Maps update=0.0017s pub=0.0053s delay=0.1361s (local map=163, WM=160)
[rtabmap-1] [INFO] [1730234045.838148980] [gary.nav.rtabmap.rtabmap]: rtabmap (305): Rate=1.00s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.0044s, Maps update=0.0017s pub=0.0003s delay=0.1195s (local map=163, WM=160)
[rtabmap-1] [INFO] [1730234046.871486519] [gary.nav.rtabmap.rtabmap]: rtabmap (306): Rate=1.00s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.0049s, Maps update=0.0019s pub=0.0003s delay=0.1205s (local map=163, WM=160)
^C[LAUNCH WARNING] [launch.launch]: caught SIGINT
[rtabmap-1] [INFO] [1730234046.960311316] [rclcpp]: signal_handler(signum=2)
[rtabmap-1] [INFO] [1730234047.013188213] [gary.nav.rtabmap.rtabmap]: Parameters are not saved (No configuration file provided...)
[rtabmap-1] tabmap.cpp:3729::process() Localization was good, but waiting for another one to be more accurate (RGBD/MaxOdomCacheSize>0)
[rtabmap-1] [ WARN] (2024-10-29 20:32:53.856) OptimizerG2O.cpp:1285::optimize() Computing marginals: vertex 233 has negative hessian index (-1). Cannot compute last pose covariance.
[rtabmap-1] rtabmap: Saving database/long-term memory... (located at /robot/generic_persistent_data/navigation/maps/test_tags_29.6/map.db)
rtabmap: 2D occupancy grid map saved.
rtabmap: Saving database/long-term memory...done! (located at /robot/generic_persistent_data/navigation/maps/test_tags_29.6/map.db, 0 MB)```

matlabbe, 2024-10-31:
I have difficulty reproducing the issue. In your second example, how long did you wait after switching the localization and killing launch file? Is it like:
```
ros2 service call /rtabmap/set_mode_localization std_srvs/srv/Empty

<< wait 5 sec >>

ctrl-c
```

brayanpa, 2025-02-27:
It turns out the issue was related to timing on our end. After making some adjustments, it never happened again.

Thanks a lot for your help!

closed 2025-02-27 by brayanpa