introlab/rtabmap_ros #1237: visualizing 3d map

taehun-ryu, 2024-11-18:
Thank you for your excellent work.

When I run ros2 launch rtabmap_examples euroc_datasets.launch.py, it seems that the octomap is visualized by default. I understand that I can disable the octomap by unchecking the corresponding checkbox(Preferences-> 3D Rendering->GridMap Assembling panel). However, I'm wondering if there is a way to disable the octomap directly through a rtabmap parameter setting or by modifying the code.

taehun-ryu, 2024-11-18:
After looking into it a bit more, I'm starting to doubt whether that gray point is an octomap. What is that gray point?

taehun-ryu, 2024-11-18:
I fully understand visualizer setting.

matlabbe, 2024-11-23:
The gray point cloud is not the octomap, but only a 3D point cloud of the map. You can disable it at the top of the 3D Rendering settings.  Uncheck these:
[image]

If you only want to see the 3D visual features of the global map and those from odometry, check those settings:
[image]