introlab/rtabmap_ros #1238: Can't build from source : WifiSignalSubNode.cpp undefined reference to rtabmap::util3d::transformPointCloud

GiannisSp, 2024-11-20:
I am trying to **build rtabmap & rtabmap-ros from source**, because I want CUDA support RTAB map docker, that **for now I'm testing it on native Ubuntu20.04**. As far as rtabmap everything works, without building errors. 
RTABmap & RTABmap-ros version 0.21.5-noetic
VTK version 9.4 ( tried 7.1 too )
PCL version 1.14.1 (tried down to 1.11) [ can't test 1.10 'cause I don't have the right GPU architecture ] 
Boost version 1.71
OpenCV version : latest 4.x
ROS : Noetic , Ubuntu 20.04 (docker) 

_**Any help would be much appreciated**. I've struggled for days on these and past problems_

**Error code :**
```
[ 66%] Linking CXX executable /home/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub
/usr/bin/ld: warning: libpcl_io.so.1.14, needed by /usr/local/lib/librtabmap_core.so.0.21.8, may conflict with libpcl_io.so.1.10
/usr/bin/ld: CMakeFiles/rtabmap_wifi_signal_sub.dir/src/WifiSignalSubNode.cpp.o: in function `mapDataCallback(boost::shared_ptr<rtabmap_msgs::MapData_<std::allocator<void> > const> const&)':
WifiSignalSubNode.cpp:(.text+0x15e2): undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/build.make:545: /home/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub] Error 1
make[1]: *** [CMakeFiles/Makefile2:6886: rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j1" failed
```
Dockerfile code 
```
FROM nvidia/cuda:12.2.0-base-ubuntu20.04
ENV DEBIAN_FRONTEND=noninteractive

ARG depfolder=/home/

WORKDIR ${depfolder}

RUN apt update -y; \
    apt install -y build-essential wget  curl python3 python3-dev libxml2 nano git \ 
                    python3 python3-pip python3-dev python3-setuptools gfortran; 

#################
##### Install CMake  22
#################
RUN apt install -y libxrandr-dev libxinerama-dev libxi-dev libssl-dev; apt remove --purge cmake; hash -r;
RUN git clone https://github.com/Kitware/CMake.git; cd CMake; ./bootstrap && make -j60 && make install; cd ..

#cuSPARSELt
RUN apt install -y libcusparselt0 libcusparselt-dev

RUN wget -q https://developer.download.nvidia.com/compute/cuda/12.2.0/local_installers/cuda_12.2.0_535.54.03_linux.run
RUN sh cuda_12.2.0_535.54.03_linux.run --toolkit --silent
ENV LD_LIBRARY_PATH=/usr/local/cuda/lib64${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
ENV PATH=/usr/local/cuda/bin${PATH:+:${PATH}}
RUN ldconfig 

# check the appropriate package based on ubuntu version (20 vs 22)
RUN wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/cuda-keyring_1.1-1_all.deb -q
RUN echo "deb [signed-by=/usr/share/keyrings/cuda-archive-keyring.gpg] https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/ /" > /etc/apt/sources.list.d/cuda.list
RUN dpkg -i cuda-keyring_1.1-1_all.deb
RUN apt-get update -q
RUN apt-get -y install cudnn-cuda-12

# #direct FWWT to cuWWT if compatible
ENV LD_PRELOAD=/usr/local/cuda/lib64/libcufft.so

# ################
# ## LAPACK install
# ################
RUN git clone https://github.com/Reference-LAPACK/lapack.git; \
    cd lapack; \
    mkdir build; \
    cd build; \
    cmake ..;\
    cmake --build . -j60 --target install

[... 128 lines cut ...]
RUN sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -
RUN apt update -y
RUN apt install ros-noetic-ros-base -y
#[inside the container-after build] 
#source /opt/ros/noetic/setup.bash
#cd catkin_ws
#catkin_make -j60;
#source devel/setup.bash

#####################
# ROS1 Rtabmap-ros 
#####################
RUN mkdir catkin_ws; cd catkin_ws; mkdir src
RUN apt install -y ros-noetic-rtabmap* ros-noetic-libg2o ros-noetic-libpointmatcher
RUN apt remove ros-noetic-rtabmap* -y; 

RUN cd catkin_ws; git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros; cd src/rtabmap_ros; git checkout 0.21.5-noetic; git pull origin 0.21.5-noetic
RUN cd catkin_ws/src; git clone https://github.com/ros-perception/vision_opencv.git; git checkout noetic; git pull origin noetic
#[inside the container-after build] 
#cd catkin_ws
#catkin_make -j60
```

_I hope you can copy-paste without typos from my part_

matlabbe, 2024-11-24:
Maybe it is related to this warning:
```
/usr/bin/ld: warning: libpcl_io.so.1.14, needed by /usr/local/lib/librtabmap_core.so.0.21.8, may conflict with libpcl_io.so.1.10
```
As you are rebuilding everything from source, uninstall libpcl* and ros-$ROS_DISTRO-pcl* before rebuilding rtabmap
```
RUN apt remove ros-$ROS_DISTRO-pcl* libpcl* -y; 
```
You may need to rebuild [pcl_ros](https://github.com/ros-perception/perception_pcl/tree/melodic-devel) from source too in case rtabmap_ros fails to find it.

GiannisSp, 2024-11-24:
Thank you for your response.
 If I try to remove `libpcl*` , it also removes the `ros-$ROS_DISTRO-pcl-conversions` ,which is vital . Haven't installed `ros-$ROS_DISTRO-pcl` and even if it get installed silently I can see that the Makefiles target my source build installation 

**My workaround** is to delete the older version `libpcl_io.so.1.10` and create a link to `libpcl_io.so.1.14` . 

It didn't solve the issue

matlabbe, 2024-11-24:
> it also removes the ros-$ROS_DISTRO-pcl-conversions and other vital packages

Yes, so you have to rebuild them from source. 

> Haven't installed ros-$ROS_DISTRO-pcl and even if it gets installed silently I can see that the Makefiles target my source build installation

The problem is that even if rtabmap is building against the pcl built from source, on runtime the wrong library version can be taken and it will just seg fault. You may verify that with `ldd` tool.

From experience, I've found that avoiding having the same libraries with different version installed at the same time saves a lot of issues down the road. Unless you really need latest PCL version, I would stick to the binaries to make your life easier. Note also rebuilding latest PCL on machine using c++17 by default can cause some issues. See https://github.com/introlab/rtabmap/issues/1385#issuecomment-2496318298 to set the right compile flags to avoid issues with GTSAM and g2o libraries using also Eigen library.

GiannisSp, 2024-12-04:
I've come with an **update** at last . 
> EDIT: just re-read original post, PCL-CUDA is not used by RTAB-Map.

So, in worst case scenario it just doesn't utilize the cuda capabilities of PCL 

**Update**:
I've succeeded building everything from source. Instead of `catkin_make` I used `catkin_make_isolated` , for some reason it stopped building without an actual warning. I decided to skip(delete) `aruco_detect` from `fiducials` package , because I don't need it and I couldn't compile it correctly due to I had to have `OpenCV <= 4.5.5` which unfortunately does not work with `CUDA 12.2`.

**What changed:**
- I made sure to remove to remove `libpcl* libopencv* libboost*` 
- Downloaded a bunch of packages found from ros-wiki (github) , _1/3 of them had to target ros1-devel ( ex melodic-devel, kinetic-devel , ros1) branches_  
-- a few I could download with `ros-noetic-<package_name> or lib<package_name>`
- Made changes to `CMakeLists` and `package.xml` to include(find) `std_srvs` in packages that couldn't find specific header files 
- Rebuild boost to actually build python_boost ( needs some extra arguments and create a user-config.jam) 
- Compiled all the packages with `OpenGL , QT and OpenCL = ON` ( instead of turning them off , or leave them in default )
- I used the master branch of rtabmap and rtabmap_ros ( instead of checkout previous noetic release ) 
- made sure I'm using `gcc/g++ v8`
   

**Question** 
If I regularly, do the following. I still do not have GPU-acellarion in nvidia-smi (I get 0% of GPU util) . I run docker with `--gpu=all`. Do I customize/add the nodes or something after the successful setup of rtabmap_ros ?
- `play stereo_outdoorsB.bag` (from ros wiki)
- `roslaunch rtabmap_demos demo_stereo_outdoor.launch`  

_I will make a full guide of current releases (in the short future)_

matlabbe, 2024-12-07:
If docker is correctly using nvidia driver, rviz and rtabmap_viz should show up in nvidia-smi:
[image]

In this [docker example](https://github.com/introlab/rtabmap/wiki/Installation#docker), we launch docker with:
```
-e NVIDIA_VISIBLE_DEVICES=all \
-e NVIDIA_DRIVER_CAPABILITIES=all \
--runtime=nvidia \
```

For rtabmap_odom and rtabmap_slam, they may use CUDA if OpenCV is built with CUDA and some of those parameters are enabled/used:
```
rtabmap --params | grep Gpu
Param: FAST/Gpu = "false"                                  [GPU-FAST: Use GPU version of FAST. This option is enabled only if OpenCV is built with CUDA and GPUs are detected.]
Param: FAST/GpuKeypointsRatio = "0.05"                     [Used with FAST GPU.]
Param: GFTT/Gpu = "false"                                  [GPU-GFTT: Use GPU version of GFTT. This option is enabled only if OpenCV>=3 is built with CUDA and GPUs are detected.]
Param: ORB/Gpu = "false"                                   [GPU-ORB: Use GPU version of ORB. This option is enabled only if OpenCV is built with CUDA and GPUs are detected.]
Param: SIFT/Gpu = "false"                                  [CudaSift: Use GPU version of SIFT. This option is enabled only if RTAB-Map is built with CudaSift dependency and GPUs are detected.]
Param: SURF/GpuKeypointsRatio = "0.01"                     [Used with SURF GPU.]
Param: SURF/GpuVersion = "false"                           [GPU-SURF: Use GPU version of SURF. This option is enabled only if OpenCV is built with CUDA and GPUs are detected.]
Param: Stereo/Gpu = "false"                                [[Stereo/OpticalFlow=true] Enable GPU version of the optical flow approach (only available if OpenCV is built with CUDA).]
Param: Vis/CorFlowGpu = "false"                            [[Vis/CorType=1] Enable GPU version of the optical flow approach (only available if OpenCV is built with CUDA).]
```

GiannisSp, 2025-04-15:
I mark this a solved because my initial problem is satisfied.

**Although I run into** troubles after the successful build. I looks like its a **versioning problems**, starting with `RVIZ`, but I'd guess it can continue with more packages. If I could know **which versions of the packages I should use** in order to run things smoothly it would help a lot

closed 2025-04-15 by GiannisSp

matlabbe, 2025-04-19:
To build rtabmap smoothly, install dependencies with `apt get install`. Don't try rebuilding latest PCL on ubuntu 20.04 for example, just use the system version directly.

Normally, rtabmap should also build with latest version of all dependencies, though we don't systematically rebuild rtabmap as soon as any dependencies update their code. We update when people try it and they find compilation issues. We however make sure that rtabmap can be built with latest official Ubuntu and ROS releases.