introlab/rtabmap_ros #1241: stereo simulation in gazebo

Dzksaigao, 2024-11-23:
Hi,
I use libgazebo-ros-camera to create a stereo camera in gazebo and related topics can be subscribed in rviz2.
but rtabmap encounters error like this:
[image]
this is the camera_info topic:
header:
  stamp:
    sec: 79
    nanosec: 33000000
  frame_id: camera_link
height: 240
width: 320
distortion_model: plumb_bob
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 277.19135641132203
- 0.0
- 160.5
- 0.0
- 277.19135641132203
- 120.5
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 277.19135641132203
- 0.0
- 160.5
- -19.403394948792545
- 0.0
- 277.19135641132203
- 120.5
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---

matlabbe, 2024-11-24:
Is it the camera_info of the right camera? What is the camera_info topic of the left camera?

Are you using gazebo-classic? You may try this plugin instead: http://classic.gazebosim.org/tutorials?tut=ros_gzplugins#Multicamera

Dzksaigao, 2024-11-25:
Yes,It's the camera_info of the right camera.
This is the left one:
header:
  stamp:
    sec: 22
    nanosec: 434000000
  frame_id: camera_link
height: 240
width: 320
distortion_model: plumb_bob
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 277.19135641132203
- 0.0
- 160.5
- 0.0
- 277.19135641132203
- 120.5
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 277.19135641132203
- 0.0
- 160.5
- -19.403394948792545
- 0.0
- 277.19135641132203
- 120.5
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false

I'm using the same plugin in ros2.
Here is the urdf file:
[image]
[image]

Dzksaigao, 2024-11-25:
> Is it the camera_info of the right camera? What is the camera_info topic of the left camera?
> 
> Are you using gazebo-classic? You may try this plugin instead: http://classic.gazebosim.org/tutorials?tut=ros_gzplugins#Multicamera

matlabbe, 2024-11-26:
Maybe you need to use `libgazebo_ros_multicamera.so` plugin instead of `libgazebo_ros_camera.so`. In the left camera_info, P[0,3] should be 0, not -19 like on the right camera_info.

closed 2024-11-27 by Dzksaigao

Dzksaigao, 2024-11-28:
> Maybe you need to use `libgazebo_ros_multicamera.so` plugin instead of `libgazebo_ros_camera.so`. In the left camera_info, P[0,3] should be 0, not -19 like on the right camera_info.

Thanks for your suggertions! I've solved this problem by using another plugin. 
Thanks!!