introlab/rtabmap_ros #1243: Map broken when use ros2_control

AnhHazz, 2024-11-23:
I'm following the tutorial with turtlebot3 model and my model. Both work fine when using libgazebo_ros_diff_drive.so 
[image]
The rqt_graph show 
[image]

But when i change to gazebo ros2_control plugin, it the map look really bad
[image]
The rqt_graph for this 
[image]

I see that the only difference between those two plugins is the odom so maybe the problem is in there. But i haven't find out how to fix. 
Tks for reading and i hope you guys help me solve this problem

matlabbe, 2024-11-24:
Which example are you using? The one with wheel odometry and RGB-D image requires relatively good odometry. If the new plugin gives very bad odometry, it won't work as expected. The rejected loop closure is because there is not enough visual features.

AnhHazz, 2024-11-24:
Hi @matlabbe , i'm using turtlebot3_rgbd launch file which mean it's using differential_drive plugin. Everything in this example work perfectly with odometry and RGBD camera.

With the ros2_control plugin i use robot_localization to fusion plugin's odom and imu and the final odom looks quiet good. But the problem still the same

Should i change to another map with more feature? But in your example there still enough features to loop closure so i think it's not cause by map

matlabbe, 2024-11-24:
Switching to a map with more visual features won't fix the odometry drifting issue. Note that odometry is looking fine with ros2_control, maybe I didn't see if there was a bad loop closure. Can you share the database with the wrong result? Maybe the covariance in the resulting odometry topic is too large.

AnhHazz, 2024-11-25:
> Switching to a map with more visual features won't fix the odometry drifting issue. Note that odometry is looking fine with ros2_control, maybe I didn't see if there was a bad loop closure. Can you [...]

https://drive.google.com/file/d/12Rz1XZm0XG331Y1bCgEiyz2lvRp2JcPD/view?usp=drive_link
Here is the link to .db file. Is that the database you mentioned? If not please tell me what to do. I'm new to this

matlabbe, 2024-12-01:
I think looks normal. The odometry is slowly drifting, which is expected on any robots. There is no loop closures because the environment is mainly featureless. That kind of environment is however pretty good for lidar proximity detections. If you cannot use lidar, you may try with turtlebot3 "house" world.

AnhHazz, 2024-12-03:
Thanks for your help. The turtlebot3_house world works as expected