introlab/rtabmap_ros #1249: /rtabmap/rgbd_sync: Did not receive data since 5 seconds!

X-lanni, 2024-12-03:
I am planning to deploy rtabmap to my robot, I have two orbbec cameras, dabai and astra pro2. The dabai camera works well, but the astra pro2 camera fails.What is the reason for this?

My setup is:

```
ubuntu20.04 ros1 noetic
Jetson nano
opencv 4.8.0
rtabmap 0.21.8
dabai and Astra pro2(OrbbecSDK_ROS1)
```

My launch file:

```xml
<launch>
  <group ns="rtabmap">

    <!-- Use RGBD synchronization -->
    <!-- Here is a general example using a standalone nodelet, 
         but it is recommended to attach this nodelet to nodelet 
         manager of the camera to avoid topic serialization -->
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_sync/rgbd_sync" output="screen">
      <remap from="rgb/image"       to="/camera/color/image_raw"/>
      <remap from="depth/image"     to="/camera/depth/image_raw"/>
      <remap from="rgb/camera_info" to="/camera/color/camera_info"/>
      <remap from="rgbd_image"      to="rgbd_image"/> <!-- output -->
      
      <!-- Should be true for not synchronized camera topics 
           (e.g., false for kinectv2, zed, realsense, true for xtion, kinect360)-->
      <param name="approx_sync"       value="true"/> 
    </node>

    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id" type="string" value="base_footprint"/>

          <param name="subscribe_depth" type="bool" value="false"/>
          <param name="subscribe_rgbd" type="bool" value="true"/>
          <param name="subscribe_scan" type="bool" value="true"/>

          <remap from="odom" to="/odom"/>
          <remap from="scan" to="/scan"/>
          <remap from="rgbd_image" to="rgbd_image"/>

          <param name="queue_size" type="int" value="10"/>
          <!--param name="topic_queue_size" type="int"  value="10"/-->

          <!-- RTAB-Map's parameters -->
          <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
          <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
          <param name="Grid/FromDepth"            type="string" value="false"/> <!-- occupancy grid from lidar -->
          <param name="Reg/Force3DoF"             type="string" value="true"/>
          <param name="Reg/Strategy"              type="string" value="1"/> <!-- 1=ICP -->
          
          <!-- ICP parameters -->
          <param name="Icp/VoxelSize"                 type="string" value="0.05"/>
          <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>
    </node>
     <node pkg="rviz" type="rviz" name="rviz" args="-d $(find myagv_navigation)/rviz/rtabmap.rviz" output="screen"/>
      
  </group>
</launch>
```

My dabai camera can run normally, and the /rtabmap/MapData data can be observed in rviz, but there is the following warning.

```bash
[... 110 lines cut ...]

          topic              rate   min_delta   max_delta   std_dev    window
=============================================================================
/camera/color/camera_info   29.97   0.02494     0.04221     0.003274   181
/camera/color/image_raw     30.01   0.02781     0.03914     0.002163   181
/camera/depth/image_raw     30.0    0.02288     0.04411     0.002663   181

          topic              rate   min_delta   max_delta   std_dev    window
=============================================================================
/camera/color/camera_info   29.99   0.02494     0.04353     0.003436   211
/camera/color/image_raw     30.0    0.02567     0.04235     0.002435   211
/camera/depth/image_raw     30.01   0.02288     0.04411     0.002525   211

          topic             rate   min_delta   max_delta   std_dev    window
============================================================================
/camera/color/camera_info   30.0   0.02494     0.04353     0.00348    241
/camera/color/image_raw     30.0   0.02567     0.04235     0.002399   241
/camera/depth/image_raw     30.0   0.02288     0.04411     0.002555   241

          topic              rate   min_delta   max_delta   std_dev    window
=============================================================================
/camera/color/camera_info   30.11   0.003144    0.04353     0.00389    272
/camera/color/image_raw     30.11   0.007947    0.04235     0.002837   272
/camera/depth/image_raw     29.99   0.02288     0.04411     0.00265    272
```

matlabbe, 2024-12-04:
Check the timestamp of these topics, maybe it is not set or it is not the same/close between all topics.

X-lanni, 2024-12-09:
Thanks for your reply, I have solved the problem. I need to modify the launch file of the Astra_pro2 camera to run it properly.

```xml
<arg name="use_hardware_time" default="false"/> 
<param name="use_hardware_time" value="$(arg use_hardware_time)"/> 
```

Because the dabai camera uses the ROS system timestamp, while the Astra_pro2 uses the hardware timestamp.

[image]
[image]

closed 2024-12-09 by X-lanni