introlab/rtabmap_ros #1250: some question about turtlebot3_sim_rgbd_scan_demo.launch.py ​

ccblog-cmd, 2024-12-03:
When I run turtlebot3_sim_rgbd_scan_demo.launch.py ​​--- vision and 2D radar fusion, rviz only shows the grid sub-map, not the 3D point cloud map. How can I set it up? In the example given by the author of RO2 humble branch, both are displayed in rviz.
This is my running result, rtabmap runs normally：
[image]
[image]
This is the author's running result：
[image]

matlabbe, 2024-12-04:
The 3D point cloud is there in your screenshot. By default the nav2 rviz config shows an orthogonal view, you can change the View to Orbit under Views panel in rviz.

In the demo screenshot, the `rtabmap_rviz_plugins::MapCloud` plugin in rviz was used.

ccblog-cmd, 2024-12-05:
> The 3D point cloud is there in your screenshot. By default the nav2 rviz config shows an orthogonal view, you can change the view to Orbit under Views panel in rviz.
> 
> In the demo screenshot, the `rtabmap_rviz_plugins::MapCloud` plugin in rviz was used.

thank you,it's ok.

closed 2024-12-07 by matlabbe