introlab/rtabmap_ros #1252: Cannot run rtabmap against rosbag

jsreveal, 2024-12-06:
Hi guys,

I am not being able to run rtabmap against recorded topics from a rosbag, while being able to run ok when live. I am using the suggestions from  #1071 but still get the following error message:
``rgbd_odometry-1] [ERROR] [1733444410.557515875] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "camera_link", TF not available at time 1733442584.769126 ``

I am running Humble on Ubuntu22, with a RealSense D455.

Launching realsense topics with the following command:
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_gyro:=true enable_accel:=true unite_imu_method:=2 enable_sync:=true align_depth.enable:=true rgb_camera.color_profile:="848,480,30" publish_tf:=false 
`

Filtering imu data with:

`ros2 run imu_filter_madgwick imu_filter_madgwick_node --ros-args --remap /imu/data_raw:=/camera/camera/imu --remap /imu/data:=/rtabmap/imu --ros-args -p use_mag:=false -p publish_tf:=false
`

Recording rosbag with the command:
`ros2 bag record   /camera/camera/color/image_raw/compressed /camera/camera/aligned_depth_to_color/image_raw/compressedDepth /rtabmap/imu /tf /tf_static`

Playing rosbag with:
`
ros2 bag play   rosbag_file  --remap /tf_static:=/tf
`

Running rtabmap with:

`ros2 launch rtabmap_launch rtabmap.launch.py     args:="-d compressed:=true"       rgb_topic:=/camera/camera/color/image_raw       depth_topic:=/camera/camera/aligned_depth_to_color/image_raw       camera_info_topic:=/camera/camera/color/camera_info       frame_id:=camera_link    wait_imu_to_init:=true    imu_topic:="/rtabmap/imu"`

Output of the bag info:

```
[INFO] [1733446201.782545134] [rosbag2_storage]: Opened database 'rosbag2_2024_12_05-23_49_41/rosbag2_2024_12_05-23_49_41_0.db3' for READ_ONLY.

Files:             rosbag2_2024_12_05-23_49_41/rosbag2_2024_12_05-23_49_41_0.db3
Bag size:          68.1 MiB
Storage id:        sqlite3
Duration:          7.918042267s
Start:             Dec  5 2024 23:49:41.797976572 (1733442581.797976572)
End:               Dec  5 2024 23:49:49.716018839 (1733442589.716018839)
Messages:          1995
Topic information: Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 1 | Serialization Format: cdr
                   Topic: /camera/camera/aligned_depth_to_color/image_raw/compressedDepth | Type: sensor_msgs/msg/CompressedImage | Count: 200 | Serialization Format: cdr
                   Topic: /camera/camera/color/image_raw/compressed | Type: sensor_msgs/msg/CompressedImage | Count: 226 | Serialization Format: cdr
                   Topic: /rtabmap/imu | Type: sensor_msgs/msg/Imu | Count: 1568 | Serialization Format: cdr
                   Topic: /tf | Type: tf2_msgs/msg/TFMessage | Count: 0 | Serialization Format: cdr

```

jsreveal, 2024-12-06:
I noticed the bag with compressed topics had not the same amount of frames for the two camera topics. Changed to the original topics but the same issue remains from rtabmap with the same message:

```
[INFO] [1733484022.279798229] [rosbag2_storage]: Opened database 'rosbag2_2024_12_06-11_15_51/rosbag2_2024_12_06-11_15_51_0.db3' for READ_ONLY.

Files:             rosbag2_2024_12_06-11_15_51/rosbag2_2024_12_06-11_15_51_0.db3
Bag size:          1.5 GiB
Storage id:        sqlite3
Duration:          11.511962344s
Start:             Dec  6 2024 11:15:51.659108893 (1733483751.659108893)
End:               Dec  6 2024 11:16:03.171071237 (1733483763.171071237)
Messages:          2996
Topic information: Topic: /rtabmap/imu | Type: sensor_msgs/msg/Imu | Count: 2308 | Serialization Format: cdr
                   Topic: /camera/camera/color/image_raw | Type: sensor_msgs/msg/Image | Count: 344 | Serialization Format: cdr
                   Topic: /camera/camera/aligned_depth_to_color/image_raw | Type: sensor_msgs/msg/Image | Count: 344 | Serialization Format: cdr

```

matlabbe, 2024-12-07:
The `/camera/camera/color/camera_info` topic is missing in the recorded bag.

For tf error, `"camera_imu_optical_frame" to frame "camera_link"` missing, it seems you launched `rs_launch.py` with `publish_tf:=false`, try with `publish_tf:=true`. You can double check with `ros2 run tf2_tools view_frames`:
[image]