introlab/rtabmap_ros #1253: Unable to colcon build rtabmap_sync package

sa32953, 2024-12-10:
Hi, I am trying to build from source as mentioned here in ros2 branch of the repo: [https://github.com/introlab/rtabmap_ros/tree/ros2](https://github.com/introlab/rtabmap_ros/tree/ros2).

I am building it on Ubuntu 22.04 on a VM with ROS2 Humble as distribution. I am unable to build a specific pakcages i.e. rtabmap_sync.
When I try to build it, it completes 21% and the system freezes every time. Here's the colcon command I am running in the workspace. 

`colcon build --symlink-install --cmake-args -DRTABMAP_SYNC_MULTI_RGBD=ON -DRTABMAP_SYNC_USER_DATA=ON -DCMAKE_BUILD_TYPE=Release --packages-select rtabmap_sync`

Any hints what could be the reason for this?

I also tried building by limiting the parallel workers using the below command. Same issue persists!

`colcon build --symlink-install --cmake-args -DRTABMAP_SYNC_MULTI_RGBD=ON -DRTABMAP_SYNC_USER_DATA=ON -DCMAKE_BUILD_TYPE=Release --packages-select rtabmap_sync --parallel-workers 2`

matlabbe, 2024-12-14:
You may ran out of RAM. Try:
```
export MAKEFLAGS="-j1"
```
before doing colcon build. Increase number of threads accordingly to your RAM.

sa32953, 2024-12-21:
Hey @matlabbe. Thanks for the reply. I will try that out. 
Do you recommend any optimal systems requirements for compiling and runtime?
Also I dont have a GPU, do you recommend running it without GPU?

matlabbe, 2024-12-22:
GPU is not required, unless you want to visualize 3D point clouds live. The underlaying visual odometry and loop closure / grpah optimization performance should be fairly the same without GPU.

With an i7, it would work as expected at runtime. For compilation, having at least 4 GB of RAM is necessary (when using `make -j1`), but having 16GB or 32 GB you can build more files in parallel.