introlab/rtabmap_ros #1255: Running rtab in ros2 humble

seongminJadenLeeunib, 2024-12-13:
I am currently trying to map 3D images using rtabmap, but the demo file does not work.

The following log appears: How can I solve this?

ros2 launch rtabmap_example realsense_d435i_stereo.launch.py
ros2 launch rtabmap_examples realsense_d435i_color.launch.py
Both have the same problem.

[rtabmap_viz-4] [WARN] [1734099294.581896098] [rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Ajusting topic_queue_size (10) and sync_queue_size (10) can also help for better synchronization if framerates and/or delays are different. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap_viz-4] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-4]    /odom \
[rtabmap_viz-4]    /camera/infra1/image_rect_raw \
[rtabmap_viz-4]    /camera/infra2/image_rect_raw \
[rtabmap_viz-4]    /camera/infra1/camera_info \
[rtabmap_viz-4]    /camera/infra2/camera_info \
[rtabmap_viz-4]    /odom_info

matlabbe, 2024-12-15:
Can you show the full log? I cannot reproduce the issue, both examples are working on my side.

SeongminJaden, 2024-12-17:
it's full logs. working in ros2 humble

matlabbe, 2024-12-21:
You may want to check if all these topics are published:
```
ros2 topic hz /camera/infra1/image_rect_raw
ros2 topic hz /camera/infra2/image_rect_raw
ros2 topic hz /camera/infra1/camera_info
ros2 topic hz /camera/infra2/camera_info 
```

SeongminJaden, 2024-12-22:
Publishing is progressing normally. Is there a topic name for rtab? I am receiving data from Intel RealSense and remapping topic names.

matlabbe, 2024-12-22:
If all those topics are published, you should not get the `rtabmap_viz` warning.

In this example, ` /odom` would be published by rtabmap's visual odometry node. Is `/odom` rate 1 Hz, 10  Hz or 30 Hz? If rtabmap node is working properly, you should have `/mapData` or `/rtabmap/mapData` topic published at 1 Hz (default). If not, you would have some warnings on the console, that you can copy paste here.