introlab/rtabmap_ros #1256: Is there any way for to do SLAM using only one monocular camera with rtabmap.

rag0nn, 2024-12-14:
Hi,
I'm beginner for SLAM. I tried to use rtabmap with monocular data but ı can't handle calculation of /odom data. Rtabmap has 3 nodes to do this: rgbd_odometry, icl_odometry and stereo_odometry, is there any way to use with monocular data these nodes with different parameters? Or is this impossible to do this without depth data.
Thanks!

My Setup:
ROS2 Humble
Ubuntu 22.04

matlabbe, 2024-12-15:
We don't provide an odometry node to do mononular visual odometry. You may search for a VIO package/library to do it.

rag0nn, 2024-12-17:
Thanks for helps!
I'm using MiDaS to calculate depth images. I provide camera info, rgb image and depth image to rgbd_odometry node but node doesn't send any data. Is there any detailed guide for this?

matlabbe, 2024-12-21:
Is `rgbd_odometry` logging something? How long is required to generate the depth image, there could be a sync issue where depth image has significant more delay than the two other topics. The `sync_queue_size` and/or `topic_queue_size` could be increased. Are depth images published with same stamp than input image?

Using these two tools on all input topics can give a better idea:
```
ros2 topic hz ...
tos2 topic delay ...
```

rag0nn, 2025-01-08:
I tried so much things. Currently, i am here:
[image]
I think first problem is extracting 954 features but odometry node dont works.
And the second i can't really understand the approach of the data synchronization. If i set approx_sync=False, not works again even not red screens appears.

matlabbe, 2025-01-13:
The depth image may not be valid, or understandable by rgbd_odometry. In the screenshot, it extracts over 900 features, but warning "20 visual features required.." is shown. It means it cannot have at least 20 features with valid depth to initialize. It is then publishing null odometry, for which rtabmap node produces an error that odometry is null (which is the case). The issue is that from the 900 features, none of them have valid depth.

The depth format should be 32FC1 in meters, or 16UC1 in mm. If the depth image is valid, maybe the error is coming from the calibration, or camera_info topic. Make sure P matrix in camera_info topic is valid and resolution matches the color image.

If rgb and depth images don't have exact same stamp, you should use approximate synchronizer.

rag0nn, 2025-02-17:
Although I haven’t fully figured out the issue, I have noticed a synchronization problem for certain reasons. Even though I provide data within the same code block (like on my previous message note3), ROS 2 seems unable to publish two consecutive messages fast enough. This happens despite both publishing and processing occurring on the same computer.

To better understand the issue, I tested RTAB-Map’s RGB-D functionality using the original Freiburg Household bag file. Then, I tested it with a filled bag file containing data from my own provider nodes and also tested it with only my provider nodes in real-time.

RTAB-Map worked correctly with the original Freiburg bag file.
It worked partially correctly with the filled bag file.
It did not work correctly with real-time publishing nodes.
I don’t fully understand why there is a difference between bag file playback and real-time publishing, but I suspect the issue is related to ROS 2 rather than RTAB-Map.

I will try running it with an RGB-D camera soon.
Thanks for your help and time! @matlabbe

closed 2025-02-17 by rag0nn