introlab/rtabmap_ros #1264: 2d Map Editor in database viewer save and view issue

lturing, 2025-01-05:
hi, just same as [issue 1173](https://github.com/introlab/rtabmap_ros/issues/1173), i update my rtabmap the latest version, commit id 2f431ce3858c39423e9280d5e4a0921580aa9079. i use my stereo mapping data rtabmap.db and [issue 1173](https://github.com/introlab/rtabmap_ros/issues/1173) provided db, after edit the optimized 2d map and save, the graph view in the rtab-map database viewer is the same as the origin, not the edited one, when using this edited db as localization mode, the 2d map is the same as the origin too.

matlabbe, 2025-01-13:
There is two things:

> after edit the optimized 2d map and save, the graph view in the rtab-map database viewer is the same as the origin, not the edited one

Do you mean in Graph View? The one in Graph View is always reconstructed from local grids, it won't show any edits. To see your changes, you should do "File->Edit optimized 2D map...", if your edits are not there too, then there is an issue.

For the localization mode, I'll need more info. There are events in localization mode than can reset the map. If you can share the database, I could launch it in localization mode to see if the changes are gone.

anath93, 2025-02-14:
@matlabbe, Hello. 
I have this same issue, where the obstacles disappear during the runtime. What I mean is, we export the map initially --> edit using GNU --> import. Load in localization mode --> somewhere in runtime the cleared obstacles show up, the question is where ? As we are in production with robot, hard to find. The only setup we have is, using tags. Tried this a few months ago, without tags I will agree the commit works but with tags, somehow the cleared obstacles come back. Is there someplace I could debug more to get more info ?

[image]
Robot sitting at charger for example, cleared this so many times but it comes back after a day or so as this causes our BT to fail eventually due to start in lethal space as we dynamically change footprint while hauling. I am latest commit, tried clearing and rebuilding workspace.

This is my [DB](https://easyupload.io/pk5k2e)

anath93, 2025-03-01:
@matlabbe Setting this Rtabmap/StartNewMapOnLoopClosure=false, solves the cleared obstacles coming back or added obstacles disappearing, maybe Odom/ResetCountdown = 1 was conflicting with it ? Will this explain or there could be another explanation her e for the global loop closure side ?

closed 2025-03-05 by lturing