introlab/rtabmap_ros #1265: Localization Jump To Random Point When Exiting Elevator

brayanpa, 2025-01-13:
**Description:**

Hi Mathieu,

I’m encountering an issue where the robot’s localization jumps to an unexpected point when exiting an elevator. Let me provide more details about the scenario:

**Setup and Behavior:**

Our robot navigates across multiple floors, switching maps when entering an elevator. The map-switching process is quick (a few seconds) as we use LiDAR-only maps. After entering the elevator, we publish on the /initialpose topic to update RTAB-Map's localization. At this stage, everything works as expected.

However, upon exiting the elevator, the localization jumps to what seems to be a random point. I suspect this point corresponds to the last saved position in the database.

**Key Observations:**

- The jump does not depend on time. For example, leaving the robot inside the elevator for several minutes still results in the localization jump immediately upon exiting.

- Logs repeatedly show the following message before the jump occurs:

```
RegistrationIcp.cpp:908::computeTransformationImpl() Transform is found (xyz=0.023405,0.240272,0.000000 rpy=0.000000,-0.000000,0.241568) but no correspondences has been found!? Variance is unknown  
(2025-01-10 22:45:52.253) Rtabmap.cpp:1321::process() Update map correction based on last localization saved in database! correction = xyz=-87.723305,-22.524700,0.000000 rpy=0.000000,-0.000000,1.795553, nearest id = 40 of last pose = xyz=2.190000,-5.140000,0.000000 rpy=0.000000,-0.000000,1.570000, odom = xyz=-3.091488,-91.526352,0.000000 rpy=0.000000,0.000000,-0.225553
```
- Inside the elevator, the robot’s LiDARs detect only a few points. The jump happens when it exits and starts seeing a significant amount of new points.
- The elevator is not mapped; it is within the map’s boundaries but marked as an unknown area.

**Additional Notes:**
This issue happens consistently with certain maps.
[MapsWJumps.zip](https://github.com/user-attachments/files/18402976/MapsWJumps.zip)

It does not occur with other maps.
[noJumps.zip](https://github.com/user-attachments/files/18402981/noJumps.zip)

Thank you for your support! Let me know if you need additional logs, configuration files, or database files.

matlabbe, 2025-01-19:
You would need to wait for rtabmap to be initialized before sending a new initial pose. If you restart the node, you can also set parameter `initial_pose` to rtabmap node to overwrite the one in the database on startup. Without restarting node, you could probably call service `load_database`, wait for the service to return successfully, then send `initialpose` topic (in map frame).

I tested all your maps with this scan-only demo to provide odom and scan (otherwise the jump won't happen till odom is revived):
```
ros2 launch rtabmap_demos turtlebot3_sim_scan_demo.launch.py localization:=true
```

I tested all your maps by dynamically changing the maps with:
```
ros2 service call /rtabmap/load_database rtabmap_msgs/srv/LoadDatabase "database_path: '~/Downloads/MapsWJumps/map5.db'"
# Send `/initialpose` from RVIZ 2d pose estimate button. Robot jumps where I clicked, as expected.

ros2 service call /rtabmap/load_database rtabmap_msgs/srv/LoadDatabase "database_path: '~/Downloads/MapsWJumps/map6.db'"
# Send `/initialpose` from RVIZ 2d pose estimate button. Robot jumps where I clicked, as expected.

ros2 service call /rtabmap/load_database rtabmap_msgs/srv/LoadDatabase "database_path: '~/Downloads/MapsWJumps/map9.db'"
# Send `/initialpose` from RVIZ 2d pose estimate button. Robot jumps where I clicked, as expected.

ros2 service call /rtabmap/load_database rtabmap_msgs/srv/LoadDatabase "database_path: '~/Downloads/noWJumps/map.original.db'"
# Send `/initialpose` from RVIZ 2d pose estimate button. Robot jumps where I clicked, as expected.
```

Are you on ros1 or ros2? I could try the same example on ros1.

brayanpa, 2025-02-27:
After extensive testing, we finally found the issue.

Our TF setup looks like this:

```
map -> odom_offset -> odom -> base_link

```

We have rtabmap configured as follows:

```
map_frame_id: map
odom_frame_id: odom_offset
```

In our case, we apply corrections to the odometry based on tag detections in the odom_offset -> odom transform. It seems that when obtaining the first localization in a map, the position calculation fails, sending the robot to an incorrect location.

We solved this by resetting the odom_offset -> odom transform every time we switch maps. However, we’re not sure if there’s a better solution by tweaking rtabmap’s configuration. If such an option exists, it would be very helpful.

Thanks a lot!

closed 2025-02-27 by brayanpa