introlab/rtabmap_ros #1267: Lidar deskewing: std::runtime_error, cannot store a negative time point in rclcpp::Time

robin-zealrobotics, 2025-01-17:
Not sure if I'm doing something wrong or whether this is a bug. So far I've not been able to find a problem with the data input that is to be deskewed.

When deskewing either with "lidar_deskewing" or as part of "icp_odometry", this error pops up.
```
terminate called after throwing an instance of 'std::runtime_error'
   what():  cannot store a negative time point in rclcpp::Time
```
This error comes right after a log statement
```[imu_filter_madgwick_node-2] [INFO] [2025-01-17 15:47:23.958] [runner_0.localisation.imu_filter]: First IMU message received.```

I'm running ros2 jazzy and I'm using a Livox mid-360 lidar. My setup (including imu) works for SLAM, as long as I don't enable the deskewing.
When doing `ros2 topic delay <imu-topic>` I get either 0 or 1ms and if I do `ros2 topic delay <lidar-topic>` I get around 100ms, both seem to be correct (lidar is set to 10Hz).
I'm using the madgwick filter to add the orientation data to the sensor (and I also bugfixed a problem in the livox driver that was putting the acceleration data in Gs instead of m/s2).
And imu_to_tf node is running with fixed-frame-id set to 'odom_imu', base-frame set to 'base_link'. And the lidar_deskewing has fixed-frame-id set to 'odom_imu'

If any more information is needed, let me know.

matlabbe, 2025-01-19:
All your steps look fine. The error may be coming from the time format used in the point cloud data. The time may be wrongly converted to a wrong data format that gives negative time.
https://github.com/introlab/rtabmap_ros/blob/cbd3088eba943b434ffc0947de2be11bdc078223/rtabmap_conversions/src/MsgConversion.cpp#L3080-L3081

We fixed a time conversion issue 2 months ago: https://github.com/introlab/rtabmap_ros/commit/e5667b280a516e1b610438d7985db67678016b33

that fix may didn't make into last binary release on ROS2 jazzy. You would have to rebuild rtabmap_ros from source if it is the issue.

robin-zealrobotics, 2025-01-20:
That indeed fixed the problem, thx!

closed 2025-01-20 by robin-zealrobotics