introlab/rtabmap_ros #1270: Sync issue in Remote mapping between RPi5 and Ubuntu Linux

rdlynx19, 2025-02-01:
Hello,
I am trying to remote mapping using an **Oak D Lite Camera.**
```
Camera Platform: Raspberry Pi5 w Ubuntu 24.04 and ROS2 Jazzy
Remote Station: System 76 Laptop w Ubuntu 24.04 and ROS2 Jazzy
```
I tried running the example `depthai.launch.py` on my laptop (remote station) and it works fine. Now, I'm trying to do remote mapping by running the camera nodes on the RPi5, and the mapping nodes on my remote laptop. Both the systems are using `rmw_zenoh_cpp` as the ROS2 DDS to discover the topics and nodes. I have verified that the topics and nodes running on the RPi are listed by running `ros2 topic list` on my remote station.

However, when I use this launch file on the RPi5 ( ideally only the camera and sync nodes should be run here)
```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    parameters=[{'frame_id':'oak-d-base-frame',
                 'subscribe_rgbd':True,
                 'subscribe_odom_info':True,
                 'approx_sync':False,
                 'wait_imu_to_init':True}]

    remappings=[('imu', '/imu/data')]

    return LaunchDescription([

        # Launch camera driver
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([os.path.join(
                get_package_share_directory('depthai_examples'), 'launch'),
               '/stereo_inertial_node.launch.py']),
                launch_arguments={'depth_aligned': 'false',
                                  'enableRviz': 'false',
                                  'monoResolution': '400p'}.items(),
        ),

        # Sync right/depth/camera_info together
        Node(   
           package='rtabmap_sync', executable='rgbd_sync', output='screen',
            parameters=parameters,
            remappings=[('rgb/image', '/right/image_rect'),
                        ('rgb/camera_info', '/right/camera_info'),
                        ('depth/image', '/stereo/depth')]),

         Compute quaternion of the IMU
        Node(
            package='imu_filter_madgwick', executable='imu_filter_madgwick_node', output='screen',
            parameters=[{'use_mag': False, 
                         'world_frame':'enu', 
                         'publish_tf':False}],
          remappings=[('imu/data_raw', '/imu')]),

        # Visual odometry
        #Node(
        #   package='rtabmap_odom', executable='rgbd_odometry', output='screen',
        #    parameters=parameters,
        #   remappings=remappings),

        # VSLAM
       #Node(
       #     package='rtabmap_slam', executable='rtabmap', output='screen',
       #    parameters=parameters,
       #      remappings=remappings,
       #    arguments=['-d']),

      # Visualization
      #  Node(
      #      package='rtabmap_viz', executable='rtabmap_viz', output='screen',
      #      parameters=parameters,
      #      remappings=remappings)
    ])

```

And this launch file on my laptop (all the mapping nodes and visualisation should be run here)
```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    parameters=[{'frame_id':'oak-d-base-frame',
                 'subscribe_rgbd':True,
                 'subscribe_odom_info':True,
                 'approx_sync':False,
                 'wait_imu_to_init':True}]

    remappings=[('imu', '/imu/data')]

    return LaunchDescription([

        # Launch camera driver
        #IncludeLaunchDescription(
        #    PythonLaunchDescriptionSource([os.path.join(
        #        get_package_share_directory('depthai_examples'), 'launch'),
        #        '/stereo_inertial_node.launch.py']),
        #        launch_arguments={'depth_aligned': 'false',
        #                          'enableRviz': 'false',
        #                          'monoResolution': '400p'}.items(),
        #),

        # Sync right/depth/camera_info together
        #Node(   
        #    package='rtabmap_sync', executable='rgbd_sync', output='screen',
        #    parameters=parameters,
        #    remappings=[('rgb/image', '/right/image_rect'),
        #                ('rgb/camera_info', '/right/camera_info'),
        #                ('depth/image', '/stereo/depth')]),

        # Compute quaternion of the IMU
        #Node(
        #    package='imu_filter_madgwick', executable='imu_filter_madgwick_node', output='screen',
        #    parameters=[{'use_mag': False, 
        #                 'world_frame':'enu', 
        #                 'publish_tf':False}],
        #    remappings=[('imu/data_raw', '/imu')]),

        # Visual odometry
        Node(
            package='rtabmap_odom', executable='rgbd_odometry', output='screen',
            parameters=parameters,
            remappings=remappings),

        # VSLAM
        Node(
            package='rtabmap_slam', executable='rtabmap', output='screen',
            parameters=parameters,
            remappings=remappings,
            arguments=['-d']),

        # Visualization
        Node(
            package='rtabmap_viz', executable='rtabmap_viz', output='screen',
            parameters=parameters,
            remappings=remappings)
    ])

```
I get these sync issues. I can see they suggest to change the queue_size parameters but I don't understand which launch file uses them.
```
[rtabmap-2] [WARN] [1738387989.642794450] [rtabmap]: rtabmap: Already running!                                                                                         
[rtabmap_viz-3] [INFO] [1738387989.644068305] [rtabmap_viz]: rtabmap_viz started.                                                                                      
[rgbd_odometry-1] [WARN] [1738387989.645072410] [rgbd_odometry]: Odometry: Already running!                                                                            
[rgbd_odometry-1] [WARN] [1738387993.465017903] [rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rost
opic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for 
the callback to be called.                                                                                                                                             
[rgbd_odometry-1] rgbd_odometry subscribed to:                                                                                                                         
[rgbd_odometry-1]    /rgbd_image                                                                                                                                       
[rtabmap-2] [WARN] [1738387993.716843672] [rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic
") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Ajust
ing topic_queue_size (10) and sync_queue_size (10) can also help for better synchronization if framerates and/or delays are different. Parameter "approx_sync" is false
, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap-2] rtabmap subscribed to (exact sync):
[rtabmap-2]    /odom \
[rtabmap-2]    /rgbd_image \
[rtabmap-2]    /odom_info
[rtabmap_viz-3] [WARN] [1738387994.644295687] [rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic
 hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpd
ate"). Ajusting topic_queue_size (10) and sync_queue_size (10) can also help for better synchronization if framerates and/or delays are different. Parameter "approx_sy
nc" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap_viz-3] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-3]    /odom \
[rtabmap_viz-3]    /rgbd_image \
[rtabmap_viz-3]    /odom_info

```
Let me know if anymore info is needed to debug this. Thanks!

rdlynx19, 2025-02-05:
Thanks @matlabbe !!
This solved the issue. However, I had to make one small change, For the `rgbd_relay`, I replaced the parameter `uncompress: True` with `compress: True`
I will now try my hands on merging two recorded maps (using different robots running rtabmap) and merging the recorded databases. 
I've looked at some of the documentation and tutorials, and noticed many of them are for ROS1. I understand the rtabmap was originally made for ROS1, and is gradually being ported for ROS2. Are there any plans for updating the tutorials to ROS2 versions? If it's possible, I would be really interested in contributing to this effort.

matlabbe, 2025-02-10:
Humm, if you put `rgbd_relay` on the remote computer, it would normally be `uncompress: True` to uncompress the compressed `rgbd_image` topic received, unless there is a bug in this logic:
https://github.com/introlab/rtabmap_ros/blob/856b2340b99809c19d86918e953c429253d9b201/rtabmap_util/src/nodelets/rgbd_relay.cpp#L121-L152
Otherwise, the nodes on the laptop will all have to decompress the images separately.

Most demos from ros1 have been ported to ROS2, see https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_demos#rtabmap_demos and https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_examples/launch

For an official documentation for ROS2, I still need to see how to enable sphinx documentation on this repo: https://github.com/introlab/rtabmap_ros/issues/1247, that would be here I would want to have a tutorial page.

matlabbe, 2025-03-02:
If you don't need to do online SLAM on the drone, you can indeed just record a rosbag and process the bag offline.

To run VSLAM on the drone, are you using PX4 flight controller? The PX4 would have an internal EKF filter with its IMUs for fast motion estimation. For outdoor drones, the EKF would also integrate GPS values to stabilize the drone in position. For indoor drones, it is possible to feed a "vision pose" to the EKF filter. That vision pose would come from visual odometry. If visual odometry cannot be computed at stable >=50Hz, the trick is to make a node republishing the last available pose at 50 Hz to PX4. That what we did [in this example](https://github.com/matlabbe/rtabmap_drone_example/blob/8cd48184ae3481628a939659f69f2c57f4e01d26/src/offboard_node.cpp#L168-L182).

rdlynx19, 2025-03-02:
Hello, thank you for the reply. I am indeed using a PX4 flight controller, and using MoCap cameras to feed visual odometry to the drone. My question however, was how do I tell the rtabmap nodes to use my vehicle's odometry, in addition to the rgbd odometry it is performing using the camera data?

matlabbe, 2025-03-04:
Would the vehicle's odometry be already a fusion of rgbd_odometry output with internal states/IMU of PX4? If so, either you provide that output pose from PX4 as an odometry topic and TF that you connect to rtabmap instead of the rgbd_odometry output topic or if only TF, set `odom_frame_id` of rtabmap node to use it. In both cases, make sure rgbd_odometry is not publishing a TF at the same time by setting `publish_tf` to false for rgbd_odometry.

In that [example](https://github.com/matlabbe/rtabmap_drone_example), we didn't feed the vehicle's odometry to visual odometry node, just the IMU of PX4, then provided the whole TF map->base_link (vo+loop closure detection) as Vision pose to PX4 (like your mocap setup it expects a global pose, not a relative one). What I describe in first paragraph doesn't include map->odom frame (odometry correction), not sure how it would be integrated in that case.

rdlynx19, 2025-03-04:
Great, this makes things more clear! Thanks @matlabbe !

closed 2025-03-15 by rdlynx19