introlab/rtabmap_ros #1273: Initialpose not taking into account frame_id?

robin-zealrobotics, 2025-02-10:
There might be something wrong with my setup, but so far I've seem to narrowed down on a problem with the 'initialpose' topic.

Setup:
- Rtabmap is configured with map_frame_id=='slam_map'
- a static transform is defined from 'map' to 'slam_map'
- using rviz I publish to initialpose and inspecting the message I see it correctly publishing the pose in 'map', and with the correct coordinates in that map
- monitoring the output on localization_pose, I however see it publishing the coordinates as sent from initialpose, but in the frame 'slam_map' instead, without transforming it accordingly

Not sure if this is intended? is 'map_frame_id' used to override the output frame and is rtabmap just ignoring the actual frame in initialpose? I would assume 'map_frame_id' is intended for lookups in the tf tree and was expecting it would transform initialpose to the 'map_frame_id' before using it.

matlabbe, 2025-02-12:
Well, it looks like we don't even check the `frame_id` set in `initialpose`, assuming it would be in same map frame set by rtabmap node:
https://github.com/introlab/rtabmap_ros/blob/ef8ec4357d76fc83351acfaff0fc31c778214f46/rtabmap_slam/src/CoreWrapper.cpp#L2527-L2537

To support `initialpose` sent in `frame_id` different than `map_frame_id` set to rtabmap, we would need a TF lookup transform to get it in `map_frame_id`.

A workaround in rviz would be to set temporary the global frame to `slam_map`, send the initial pose, then set back to `map`.

matlabbe, 2025-03-02:
I added support of `initialpose` sent in any frame in the commit above. If the pose cannot be transformed, an error message will be shown. If the `frame_id` is empty, there will be a warning, assuming it is already in map frame.

closed 2025-03-02 by matlabbe