introlab/rtabmap_ros #1274: Localization using Monocular RGB Camera

zayedreply, 2025-02-11:
Hi, I am interested in doing the localization using a monocular rgb camera. I have mapped the area using a realsense d455 rgbd camera. I used the rgb + depth + imu to make the area. However i am not managing to do the localization with the rgb camera only. 

Is it possible to do so? For the mapping I have used the ros2 launch file provided, with just the topic names changed. 

Thanks in advance

matlabbe, 2025-03-01:
Can you share a rosbag of the rgb frames (+ camera_info + tf + tf_static) along the database you are trying to localize on?

If you are using the same RGB camera used during mapping, it should be roughly working as well as using RGB+Depth in localization mode. Are you able to localize with using same input sensors than mapping?

zayedreply, 2025-07-21:
Hi @matlabbe. Thanks a lot for your help and I am very sorry for the late reply. My problem is similar to that in the [issue](https://github.com/introlab/rtabmap/issues/1395) mentioned here
I have a warehouse/office where i am walking around and mapping using a handheld realsense. To increase the performance, I have installed apriltags all over the place. The mapping step yeilds very good results. I want then to localize a drone which has only a front facing camera in this environment.

I have noticed that in localization mode. The pose jumps basically until an apriltag is infront of the camera. otherwise there is no localisation.

I have used SIFT features for kp/DetectorStrategy as suggested, with no improvement. The drone camera has a different resolution and fov as the realsense.

here are my launch files

[rtabmap_map.txt](https://github.com/user-attachments/files/21344521/rtabmap_map.txt)
[rtabmap_rgb_loc.txt](https://github.com/user-attachments/files/21344522/rtabmap_rgb_loc.txt)

I am guessing i am doing something wrong with the parameters.

Thanks in advance