introlab/rtabmap_ros #1279: IMU and map frame are not aligned

rdlynx19, 2025-02-16:
Hi @matlabbe
Thanks for your help on #1270 . As mentioned in the previous issue, I'm running the example `depth_ai.launch.py` file by splitting it into two parts for my setup. I am using an Oak D Lite camera. 
```
Camera Platform: Raspberry Pi5 w Ubuntu 24.04 and ROS2 Jazzy
Remote Station: System 76 Laptop w Ubuntu 24.04 and ROS2 Jazzy
```
However, now that all the nodes seem to be running fine, I have observed that the camera frame and the map frame are not aligned. This is causing the point cloud to appear in the incorrect direction in rviz. I also think this is causing the 2d graph to appear as completely occupied (black grids).
Attached are some screenshots.

[image]
[image]

Since I'm not publishing any additional transforms, I believed these should have been aligned by default. I am unaware if there is any specific parameter I have to modify to do this. Any help regarding this would be appreciated. Thanks!

rdlynx19, 2025-02-19:
Update: I seem to have found a temporary fix for this problem by modifying the camera urdf. This changed the static transform between the camera base frame and the camera imu frame. However, I dont think this is the correct way to do this. Thoughts?

matlabbe, 2025-03-02:
When running [depthai.launch.py](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/depthai.launch.py), you have to set the right `camera_model` matching your sensor. This will set the [static transforms](https://github.com/luxonis/depthai-ros/tree/humble/depthai_descriptions/urdf/models) between the base frame of the camera, the optical frames and the imu frame. It is possible that the IMU frame being wrong (I've seen this on one camera, don't remember which model).

matlabbe, 2025-03-04:
for the IMU orientation, you may also read this comment: https://github.com/introlab/rtabmap/pull/1458#issuecomment-2692648223

closed 2025-03-15 by rdlynx19