introlab/rtabmap_ros #1283: how to automatically relocalization instead of publish initpose manually?

lturing, 2025-02-24:
hi, thanks for your great work. i want to use rtabmap to mapping continually, not mapping mode firstly and location mode latter, because the surrounding environment is dynamic. so if i want to mapping firstly, and latter load db file and continually mapping, and the start point may be not around the latest stopping point, so i need automatically relocalize the pose of the car in the built map.

matlabbe, 2025-03-04:
If the robot is restarting somewhere in the environment contained in the map, it could globally re-localize, at the condition that the robot moves close to a node in the map's graph with same orientation (if not 360 FOV) and that same visual features can be extracted (could be harder in environment with varying illumination, [unless the visual features are already robust to that](https://arxiv.org/abs/2103.03827)). In some cases, we need to teleop or have a behavior from the robot to move randomly till it localizes. If the robot is started outside the map, it won't be able to re-localize.

lturing, 2025-03-05:
many thanks

closed 2025-03-05 by lturing