introlab/rtabmap_ros #1285: CI failing when building humble-latest

matlabbe, 2025-03-02:
See https://github.com/introlab/rtabmap_ros/actions/runs/13612255207/job/38050924306

```
 #13 250.7 CMake Error at /opt/ros/humble/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
#13 250.7   Package 'rcutils' exports the library 'rcutils' which couldn't be found
#13 250.7 Call Stack (most recent call first):
#13 250.7   /opt/ros/humble/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
#13 250.7   /opt/ros/humble/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
#13 250.7   /opt/ros/humble/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
#13 250.7   /opt/ros/humble/share/rosidl_typesupport_fastrtps_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
#13 250.7   /opt/ros/humble/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_cConfig.cmake:41 (include)
#13 250.7   /opt/ros/humble/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
#13 250.7   /opt/ros/humble/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
#13 250.7   /opt/ros/humble/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
#13 250.7   /opt/ros/humble/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
#13 250.7   CMakeLists.txt:16 (find_package)
#13 250.7 
#13 250.7 
#13 250.7 ---
#13 250.7 Failed   <<< rtabmap_msgs [23.5s, exited with code 1]
#13 251.0 
#13 251.0 Summary: 1 package finished [29.5s]
#13 251.0   1 package failed: rtabmap_msgs
#13 251.0   1 package had stderr output: rtabmap_msgs
#13 251.0   11 packages not processed
```

aslamahrahman, 2025-03-25:
i am having the same issue when trying to build even just a package containing only messages

matlabbe, 2025-03-31:
This issue doesn't seem happening anymore, but humble build still fail on that error, but only in CI: https://github.com/introlab/rtabmap_ros/issues/1288

I think I solved the "rcutils" issue above by re-generating the base image without cache. I think new Humble binaries were released and our base image was not updated, and in this repo we were installing new ros packages that were using latest humble version, so that could have been a mismatch between old and new humble version.

In your case, if you are building natively, make sure you have all humble binaries updated:
```
sudo apt upgrade ros-humble*
```

closed 2025-03-31 by matlabbe