introlab/rtabmap_ros #1289: Parameter optimization for mapping

ashBabu, 2025-03-11:
Hi 
Ubuntu 24 and ROS 2 Jazzy
My aim is to map an environment with sparse features and deploy the algorithm on either a NUC (for now) or on an edge device ultimately. So I am trying to fuse visual odometry ( intel realsense d400) and 2D lidar (okdo) and an IMU using `robot_localization` ros package which publishes the `odom-base_link` transform. 

I have setup the workflow as follows using a [bringup.launch.py](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/bringup.launch.py)
* Launch okdo lidar
* Launch [rtabmap_icp_odometry](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/rtabmap_icp_odom.launch.py) to convert the `/okdo/scan` to odometry
* Launch [realsense_camera2](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/realsense_cam.launch.py), [imu_filter_madgwick](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/imu_filter_madgwick.launch.py) to get an imu msg with orientation
* Launch [rtabmap rgbd odom](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/rtabmap_rgbd_odom.launch.py)
* Launch [robot_localization](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/localization.launch.py)
* And finally launch [rtabmap slam](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/rtabmap_slam.launch.py)

The config files are available in [config](https://github.com/ashBabu/Utilities/tree/master/rtabmap/config)

I hold the camera and lidar and walk around in my home to generate the  `.db` file which is available [here](https://www.dropbox.com/scl/fi/v3ol3uckm524zg86hpmq2/slam_2025-03-11_13-57-52.db?rlkey=30mjs6miehfvkev3tv0h3qmd0&st=yyixy0f8&dl=0).

I have carried out these tests several times, yet I dont get a good map. I have tried changing the parameters by reading through the [Paramerters.h](https://github.com/introlab/rtabmap/blob/jazzy-devel/corelib/include/rtabmap/core/Parameters.h) file, yet I think I am far away from getting a good map. 

Could you please suggest some steps to get a better map? I can add more information if I have missed any. 
I have noticed that the mapping is better(visually) on my laptop (much powerful) than on my NUC. How can I tune it for NUC

Thanks

[image]

matlabbe, 2025-03-19:
The RTAB-Map's parameters are case-sensitive, you cannot set them like this
https://github.com/ashBabu/Utilities/blob/7d324644ce1114f28b515c6a9ab69c513250225d/rtabmap/launch/rtabmap_icp_odom.launch.py#L28-L48

See `rtabmap --params` for the exact case.

For rtabmap node, I would set the same ICP parameters you set for icp_odometry + `'Reg/Strategy': '1'` and `'Reg/Force3DoF': 'true'`.

Otherwise, I kinda fixed the database for the lidar scans with:
```
rtabmap-reprocess  \
   --Reg/Strategy 1 \
   --Reg/Force3DoF true  \
   --RGBD/NeighborLinkRefining true  \
   --Icp/MaxCorrespondenceDistance 1  \
   --Icp/MaxTranslation 0.5  \
   slam_2025-03-11_13-57-52.db output.db
```
Before:

[image]

After:

[image]

Is your 2d lidar always perfectly horizontal? It seems to be titled sometimes and I think this is caused the slight misalignment in the image above.

For the realsense data, the URDF or static transform between the lidar/base link and the camera link is wrong. This is how the systems thinks they are oriented:

[image]
[image]

The camera should be tilted and the alignment with lidar could be better.

You could also remove [rtabmap rgbd odom](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/rtabmap_rgbd_odom.launch.py) and [robot_localization](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/localization.launch.py) nodes first. Normally with icp_odometry should have already a pretty accurate map for that kind of environment (assuming the lidar is always parallel to ground).

ashBabu, 2025-03-19:
Thank you @matlabbe : Indeed there is a difference in the transforms. I was trying to approximate 2D motion by hand held walking. I have corrected the case sensitive params of icp odometry and removed robot localization and rgbd odometry. 

Now I am working on a simulated robot trying to map a building. Basically what I want to do now is to use icp_odometry and register the Pointcloud from the realsense camera to construct a 3D map of the environment. The actual use case is an environment with sparse features where I think rgbd odometry may fail although I have not explored much into infra red odom.

The updated launch files are [icp_odometry](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/rtabmap_icp_odom.launch.py) and [rtabmap_slam](https://github.com/ashBabu/Utilities/blob/master/rtabmap/launch/rtabmap_slam.launch.py)
The 2D Occupancy grid looks very accurate but the 3D environment is really bad. 

[image]

[image]

The camera is modeled as in [here](https://github.com/ashBabu/Utilities/blob/master/rtabmap/model/model.sdf#L40-L80)
The db is available [here](https://www.dropbox.com/scl/fi/zehxcx98rm8tm5m397ol5/slam_2025-03-19_14-27-34.db?rlkey=wp4ny4k5x3g4v1jpfp3ql9t34&st=qt28pv2n&dl=0)

I am using `rtabmap-databaseViewer-->Edit--->View 3D Map` to create the above image
Please can you point to where I am going wrong. Thanks

matlabbe, 2025-03-20:
I think there is a scale issue coming from the depth values in the depth image, or the focal length in camera calibration is wrong:

[image]

ashBabu, 2025-03-24:
Thanks. I have just removed the calibration params in the gazebo camera sdf and it works well. Closing this