introlab/rtabmap_ros #1290: Question regarding rtabmap handling of time

Teledyne-MV-amoshkov, 2025-03-14:
I'm using ros2 rtabmap to do SLAM using a bag that plays rgdb data.
I am unclear about how time is handled in rtabmap.

I'm playing the bag in a loop (just for convenience to understand the process flow)
When running the rtabmap.launch file, I get time related messages like:
- message1 - [rgbd_odometry]: The time difference between rgb and depth frames is high (diff=0.100601s, rgb=1741900345.572263s, depth=1741900345.672865s)
or 
- message2 - [tf2_buffer]: Detected jump back in time. Clearing TF buffer.

I see various use cases:
- usecase1: playing the bag with --clock
  - the bag sets a clock server and the time is published in the /clock topic
    Is the time set from the system time when playing the bag, which 
  - nodes with "use_sim_time=True" listen to the /clock topic and get the time from the /clock topic 
- usecase2: playing the bag without --clock
  - the bag does not emit any /clock topic 
  - nodes with "use_sim_time=False" get the time from the system time 

the time in usecase1, usecase2 refer to the time that the message arrives, correct?
To run rtabmap in the most controlled manner, is there a benefit to using the simulated time or the system time?
Regardless of this time, rtabmap looks into the header timestamps (sec, nanosec) of the messages themselves, correct?

To prevent message1 I can set the parameters
- approx_sync = True
- approx_sync_max_interval - increase it to allow more tolerance between time differing messages

Regarding message2, I don't know if the "Detected jump back in time" comes from the time the message arrives or the header timestamp?
I added a clock_republisher script that listens to the /clock topic and generates an /adjusted_clock that is monotonically increasing.
rtabmap maps /clock to /adjusted_clock so I should not see this message if the reason is the time of message arriving (not the header timestamp).
Is there a way to get details of the time values and specific tf message that causes this message?

Thanks

matlabbe, 2025-03-16:
You cannot loop back a rosbag with rtabmap. rtabmap checks the header in the topics and if time jumps back in the past, the frames are ignored.

So assuming you run only one time the bag, the next thing is that you should use `--clock` with `use_sim_time:=true`, in particular if the nodes are using `rclcpp::Node::now()`. rtabmap uses this only to check if it received messages in the past 5 seconds (and show up a warning message), though if you omit `--clock` and `use_sim_time:=true`, rtabmap would still work using stamps from the topic's header even if the warning messages about "not received any data in past 5 seconds" are shown. To get rid of these warnings messages, use `--clock` with `use_sim_time:=true` set to all nodes.

Teledyne-MV-amoshkov, 2025-03-17:
rtabmap 
explain the parameter approx_sync_max_interval