introlab/rtabmap_ros #1291: "process" on a separate thread

sandeepdutta, 2025-03-18:
Hello Mathieu,

I am trying to reduce the "synchronization" delay . I noticed that all callbacks including "processAsync"  are assigned to a
"mutually exclusive" callback group, this would imply that "commonMultiCameraCallback" & "processAsync" will run
sequentially . I have created an experimental implementation which runs "process" on a separate thread , it communicates  with "commonMultiCameraCallback" using a double buffer. This has improved the latency significantly which means I can run the
robot faster .
Do you see any drawback / problem with this approach ? The code is in my repo https://github.com/sandeepdutta/rtabmap_ros/tree/ros2
I can create a "Pull" request if you think this is worth considering.

Best Regards
Sandeep

matlabbe, 2025-03-19:
Could be related to https://github.com/introlab/rtabmap_ros/pull/1214

I am not sure to see the difference from https://github.com/introlab/rtabmap_ros/compare/ros2...sandeepdutta:rtabmap_ros:ros2 than using a timer thread called in a different callback group. `commonMultiCameraCallback` and `processAsync` are not in the same callback group, they should be processed in parallel as we are also using a `rclcpp::executors::MultiThreadedExecutor` to run the node, unless I messed up the implementation. If you use rtabmap in a Composable node, then if the executor is not mutli-threaded, the callbacks will be indeed processed in sequence.

The reason why we used a Timer (that we could associate a callback group) instead of a thread is to make sure that when rtabmap is processing data, all services (associate with same callback group) cannot be executed at the same time (an issue I pointed for ros1 implementation https://github.com/introlab/rtabmap_ros/issues/1054#issuecomment-2354745697). Only the message filters callbacks (like `commonMultiCameraCallback`) are processed in parallel, and they can only exchange data to main process by the Timer interface.

sandeepdutta, 2025-03-21:
Hello Mathieu,

Thank you for your response and the link, now I understand the philosophy of the design. I checked the code,
your implementation has no flaws, however I think there is a significant delay (100ms) delay inside ROS (rclcpp)
from the time syncTimer_->reset() is called and the timer is actually invoked.  I am loading rtabmap_slam::CoreWrapper
into a "component_container_mt".
I have included the timing logs below, if you see the difference in Hz of /localization_pose you will see there is an 
average difference of about 100ms , the timing reported by rtabmap does not seem to be to different (not sure why).
Here are the changes I made.
a) added parameter processInThread_ (= True) will use Double buffer to send SyncData_ to a dedicated thread 
(processAsyncThread) , =False will reset the timer and use the existing method.
b) added processAsynThread.
c) in all callbacks in  processingCallbackGroup_ I have locked and unlocked rtabmapMutex_ to prevent  what you
mentioned in the #1054 (comment).
All changes are contained in CoreWrapper.h & CoreWrapper.cpp. The code is in my Repo (link above).
I will continue testing to see all works. Would greatly appreciate any feedback you have.

Best Regards
Sandeep

Hardware is NVIDIA Jetson ORIN AGX 64GB 12-ARM Cores.
admin@jeevesrobot:/workspaces/isaac_ros-dev$ echo ## Running on Dedicated thread

admin@jeevesrobot:/workspaces/isaac_ros-dev$ ros2 topic hz /localization_pose 
average rate: 4.752
        min: 0.168s max: 0.335s std dev: 0.05413s window: 7
average rate: 5.064
        min: 0.168s max: 0.335s std dev: 0.04270s window: 13
average rate: 5.171
        min: 0.167s max: 0.335s std dev: 0.03675s window: 19
average rate: 5.211
        min: 0.167s max: 0.335s std dev: 0.03261s window: 25
average rate: 5.170
        min: 0.167s max: 0.335s std dev: 0.03603s window: 31
average rate: 5.244
        min: 0.167s max: 0.335s std dev: 0.03363s window: 37

[component_container_mt-13] [INFO] [1742580330.833931552] [rtabmap]: rtabmap (2651): Rate=0.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.1734s, Maps update=0.0034s pub=0.0002s delay=0.5463s (local map=1157, WM=1157)
[component_container_mt-13] [INFO] [1742580331.005077487] [rtabmap]: rtabmap (2652): Rate=0.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.1681s, Maps update=0.0021s pub=0.0001s delay=0.4505s (local map=1157, WM=1157)
[component_container_mt-13] [INFO] [1742580331.171600692] [rtabmap]: rtabmap (2653): Rate=0.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.1631s, Maps update=0.0026s pub=0.0001s delay=0.3502s (local map=1157, WM=1157)
[component_container_mt-13] [INFO] [1742580331.353683577] [rtabmap]: rtabmap (2654): Rate=0.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.1791s, Maps update=0.0022s pub=0.0001s delay=0.3654s (local map=1157, WM=1157)
 -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

admin@jeevesrobot:/workspaces/isaac_ros-dev$ echo ## Running on Timer thread

admin@jeevesrobot:/workspaces/isaac_ros-dev$ ros2 topic hz /localization_pose
average rate: 2.931
        min: 0.271s max: 0.435s std dev: 0.06241s window: 4
average rate: 3.572
        min: 0.203s max: 0.435s std dev: 0.07133s window: 9
average rate: 3.529
        min: 0.203s max: 0.435s std dev: 0.06432s window: 13
average rate: 3.598
        min: 0.200s max: 0.435s std dev: 0.06966s window: 17
average rate: 3.599
        min: 0.200s max: 0.435s std dev: 0.07182s window: 21

[component_container_mt-13] [INFO] [1742580573.277878154] [rtabmap]: rtabmap (2489): Rate=0.00s, Limit=0.000s, Conversion=0.0030s, RTAB-Map=0.3080s, Maps update=0.0034s pub=0.0002s delay=0.4241s (local map=1157, WM=1157)
[component_container_mt-13] [INFO] [1742580573.534828828] [rtabmap]: rtabmap (2490): Rate=0.00s, Limit=0.000s, Conversion=0.0009s, RTAB-Map=0.1784s, Maps update=0.0023s pub=0.0003s delay=0.4141s (local map=1157, WM=1157)
[component_container_mt-13] [INFO] [1742580573.774054290] [rtabmap]: rtabmap (2491): Rate=0.00s, Limit=0.000s, Conversion=0.0012s, RTAB-Map=0.2004s, Maps update=0.0024s pub=0.0001s delay=0.6199s (local map=1157, WM=1157)
[component_container_mt-13] [INFO] [1742580574.021253507] [rtabmap]: rtabmap (2492): Rate=0.00s, Limit=0.000s, Conversion=0.0015s, RTAB-Map=0.1923s, Maps update=0.0023s pub=0.0001s delay=0.6669s (local map=1157, WM=1157)

sandeepdutta, 2025-03-29:
Hi Mathieu,

Thank you for spending the time to do this detailed analysis, it explains the 100ms difference I see in my system.
I have two cameras generating @ 15 FPS, after synchronization the image rate is at  ~10Hz. If we are skipping frames
then **process** would have to potentially wait ~100ms for the next set of synchronized frames to arrive, if we are using the
double buffer method then we do not need to wait , albeit at the price of compute.
You ask the important question **what is the best ?**, for me the answer is **it dependents**. In systems like mine where
you are dealing with multiple discrete sensors that have large synchronization delays perhaps the double buffer is
better. I understand your concern about **wasted** compute required during buffering  , I am experimenting with 
a solution for that .... we dynamically adjust the number of frames we buffer vs a number we skip based on the time
**process** takes, if process takes more time then we skip more .. adjust the number dynamically.
I will keep you updated about the progress of this experiment, please let me know your thoughts, or if you have alternate
solutions that might decrease the "wasted" time.

Thank you & best regards
Sandeep

sandeepdutta, 2025-05-03:
Hello Mathieu,

A quick update, I continue to run my setup with this change, so far all is working ok. I have done some analysis,
it looks like in my setup we are throwing away between 0-2 frames depending on RTABMAP processing time, the
localization_pose frequency stays between 4.5 - 5 Hz. I could increase the speed of my robot. 
Best Regards
Sandeep

matlabbe, 2025-05-10:
Thank you for the feedback, I started two months ago a [branch](https://github.com/introlab/rtabmap_ros/pull/1311) to implement the double buffer. I needed to do some refactoring to implement it in all topic callbacks, so it is a work in progress. Hopefully soon I will have time to finish that PR and make it official.