introlab/rtabmap_ros #1299: Outdoor mapping problem

zhangle0803, 2025-04-15:
Recently, when I was using realsensed415 for outdoor mapping, the rtabmap algorithm was affected by the lighting, resulting in poor mapping effect. There were many black spots (obstacles) on the raster map that did not exist originally. How should I solve this problem? Thank you

[image]
[image]
[image]

zhangle0803, 2025-04-15:
The first picture is the construction under cloudy conditions, and the second picture is the result of the construction under sunny conditions. It can be seen that there are many obstacles that should not exist in the middle of the second picture. How should I solve this problem? Thank you all

matlabbe, 2025-04-19:
Duplicate to https://github.com/introlab/rtabmap/issues/1484

closed 2025-04-19 by matlabbe