introlab/rtabmap_ros #1301: How to display the octree map in the Turtlebot3 Nav2, 2D LiDAR and RGB-D SLAM demo

ccblog-cmd, 2025-04-16:
I can run the Turtlebot3 Nav2, 2D LiDAR and RGB-D SLAM demo project normally, as shown below:

[image]

[image]

[image]
Related topics:
cqx@Hp:~/rtabMap_ros2_ws$ ros2 topic list
/behavior_server/transition_event
/behavior_tree_log
/bond
/bt_navigator/transition_event
/camera/camera_info
/camera/cloud
/camera/depth/camera_info
/camera/depth/image_raw
/camera/ground
/camera/image_raw
/camera/obstacles
/camera/points
/clicked_point
/clock
/cloud_ground
/cloud_map
/cloud_obstacles
/cmd_vel
/cmd_vel_nav
/cmd_vel_teleop
/controller_server/transition_event
/cost_cloud
/diagnostics
/disparity/camera_info
/disparity/image
/downsampled_costmap
/downsampled_costmap_updates
/evaluation
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_updates
/global_costmap/footprint
/global_costmap/global_costmap/transition_event
/global_costmap/published_footprint
/global_costmap/voxel_marked_cloud
/global_path
/global_path_nodes
/global_pose
/goal
/goal_node
/goal_out
/goal_pose
/goal_reached
/gps/fix
/grid_prob_map
/imu
/info
/initialpose
/joint_states
/labels
/landmark_detection
/landmark_detections
/landmarks
/local_costmap/clearing_endpoints
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_updates
/local_costmap/footprint
/local_costmap/local_costmap/transition_event
/local_costmap/published_footprint
/local_costmap/voxel_grid
/local_costmap/voxel_marked_cloud
/local_grid_empty
/local_grid_ground
/local_grid_obstacle
/local_path
/local_path_nodes
/local_plan
/localization_pose
/map
/mapData
/mapGraph
/mapOdomCache
/mapPath
/map_updates
/marker
/mobile_base/sensors/bumper_pointcloud
/octomap_empty_space
/octomap_global_frontier_space
/octomap_grid
/octomap_ground
/octomap_obstacles
/octomap_occupied_space
/odom
/parameter_events
/particle_cloud
/performance_metrics
/plan
/plan_smoothed
/planner_server/transition_event
/preempt_teleop
/proj_obstacles
/received_global_plan
/robot_description
/rosout
/rtabmap/republish_node_data
/scan
/smoother_server/transition_event
/speed_limit
/tf
/tf_static
/transformed_global_plan
/user_data_async
/velocity_smoother/transition_event
/waypoint_follower/transition_event
/waypoints

matlabbe, 2025-04-19:
You would need to set `'Grid/3D': 'true'` here: 
https://github.com/introlab/rtabmap_ros/blob/f5a3e487056e399b834971ef6a64126843666f7f/rtabmap_demos/launch/turtlebot3/turtlebot3_rgbd_scan.launch.py#L41

Then if you install `ros-$ROS_DISTRO-octomap-rviz-plugins`, you could show the octomap in rviz with ColorOccupancyGrid plugin.

Note that the actual octomap topic looks missing, here the ones you should see:
```
ros2 topic list | grep octomap
/octomap_binary
/octomap_empty_space
/octomap_full
/octomap_global_frontier_space
/octomap_grid
/octomap_ground
/octomap_obstacles
/octomap_occupied_space
```
In particular, `/octomap_full` and `/octomap_bin` are octree topics. If you built rtabmap and rtabmap_ros from source, make sure to install `liboctomap-dev` before building these projects.
```
sudo apt install liboctomap-dev ros-$ROS_DISTRO-octomap-msgs ros-$ROS_DISTRO-octomap-rviz-plugins
```

matlabbe, 2025-11-01:
If you have [ROS2](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html) installed.
```
sudo apt install \
   ros-$ROS_DISTRO-rtabmap-demos \
   ros-$ROS_DISTRO-turtlebot3-gazebo \
   ros-$ROS_DISTRO-depthimage-to-laserscan

ros2 launch rtabmap_demos turtlebot3_sim_rgbd_fake_scan_demo.launch.py
```