introlab/rtabmap_ros #1310: 3D map drifting on NovaCarter with RTAB-Map

Shailesh-Pawar-12, 2025-05-08:
I've set up RTAB-Map on the NovaCarter robot (ROS 2), but the 3D map is drifting noticeably — especially visible in the bottom-right corner of the screen in the attached screenshot. There are no errors in the console.

Attaching the launch file I'm using. Any idea what could be causing the drift? 

[image]
[image]

[rtabmap_slam3d.launch.py.txt](https://github.com/user-attachments/files/20101786/rtabmap_slam3d.launch.py.txt)

matlabbe, 2025-05-08:
The images should be rectified. There is an example here: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_demos/launch/isaac/isaac_sim_vslam_demo.launch.py

Shailesh-Pawar-12, 2025-05-09:
@matlabbe Thank you for your reply!

I'm not using VLSAM for the Nova Carter, as mentioned in your example.
Instead, I have a 3D LiDAR, an RGB-D camera, and an IMU.
Could you please share a reference launch file or example configuration that uses these sensors with RTAB-Map?

matlabbe, 2025-05-12:
I don't have example with 3D LiDAR on isaac simulator. Last time I tried it I was not satisfied with how they simulated a 3D LiDAR with 3 or 4 wide angle cameras. However, we have a Gazebo demo with similar sensors: https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_demos#clearpath-husky-nav2-3d-lidar-and-rgb-d-slam, in particular that launch file [husky_slam3d.launch.py](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_demos/launch/husky/husky_slam3d.launch.py) that could be included in your setup.