introlab/rtabmap_ros #1316: rtabmap_slam binary from release not built with nav2_msgs support (use_action_for_goal  parameter not available)

mariamercade, 2025-05-27:
Hello,

I'm using the official `rtabmap_ros` release binaries for **ROS 2 Jazzy**, installed via:

```
sudo apt install ros-jazzy-rtabmap-ros
```

When I try to enable the `use_action_for_goal` parameter for the `rtabmap` node (from the `rtabmap_slam` package), I get the following error at runtime:

```
[rtabmap-2] [ERROR] [1748359373.612831275] [rtabmap.rtabmap]: rtabmap: Cannot enable use_action_for_goal because rtabmap_slam is not built with nav2_msgs support.
```

Am I doing something wrong? If not, would it be possible to include `nav2_msgs` support in the release builds so that `use_action_for_goal` works out of the box?

Thank you!

mariamercade, 2025-07-01:
Hi, @matlabbe! If this change is already included in the latest ROS 2 Jazzy sync, then the issue can be closed. Thanks!

matlabbe, 2025-07-02:
Yeah, the new version has been released https://repo.ros2.org/status_page/ros_jazzy_default.html?q=rtabmap, closing

closed 2025-07-02 by matlabbe