introlab/rtabmap_ros #1321: Understanding RTAB's capabilities with just 3D Lidar

Edwardius, 2025-06-05:
Hello, 

I've been experimenting with RTAB and its ability to handle two lidars, and I had the following questions:
- Once a map is made, is it possible to determine the robot's location in the map on startup?
- Does an initialpose need to be set? or can RTAB match the current lidar scans with the past map to get a global pose estimate?

Here's a sample launchfile: 

```
# Description:
#   In this example, we will record ALL lidar scans from 2 lidars. An IMU or low latency odometry is required for this example.
#
# Example:
#   Launch your lidar sensors
#     In this example, we assume the lidar topics have a frame_id linked to same parent (e.g., base_link) and
#     the extrinsics are known (URDF) and/or already calibrated.
#
#   Launch your IMU sensor, make sure TF between lidar/base frame and imu is already calibrated.
#     In this example, we assume the imu topic has
#     already the orientation estimated, if not, you can launch
#     imu_filter_madgwick_node (with use_mag:=false publish_tf:=false)
#     and set imu_topic to output topic of the filter.
#
#   If a camera is used, make sure TF between lidar/base frame and camera is
#     already calibrated. To provide image data to this example, you should use
#     rtabmap_sync's rgbd_sync or stereo_sync node.
#
#   Launch the example by adjusting the lidar topics, imu topic and base frame:
#   $ ros2 launch rtabmap_examples lidar3d.launch.py lidar1_topic:=/lidar1/velodyne_points lidar2_topic:=/lidar1/velodyne_points imu_topic:=/imu/data frame_id:=base_link

from launch import LaunchDescription, LaunchContext
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

# For description
from launch.actions import GroupAction
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from pathlib import Path
from ament_index_python.packages import get_package_share_directory

def launch_setup(context: LaunchContext, *args, **kwargs):

  frame_id = LaunchConfiguration('frame_id')

  external_odom_frame_id =  LaunchConfiguration('external_odom_frame_id').perform(context)

  rgbd_image_topic = LaunchConfiguration('rgbd_image_topic')
  rgbd_image_used =  rgbd_image_topic.perform(context) != ''

  lidar1_topic = LaunchConfiguration('lidar1_topic')

  lidar2_topic = LaunchConfiguration('lidar2_topic')

  voxel_size = LaunchConfiguration('voxel_size')
  voxel_size_value = float(voxel_size.perform(context))

  use_sim_time = LaunchConfiguration('use_sim_time')

  localization = LaunchConfiguration('localization').perform(context)
  localization = localization == 'true' or localization == 'True'

  deskewing_slerp = LaunchConfiguration('deskewing_slerp').perform(context)
  deskewing_slerp = deskewing_slerp == 'true' or deskewing_slerp == 'True'

  # Rule of thumb:
  max_correspondence_distance = voxel_size_value * 10.0

  shared_parameters = {
[... 163 lines cut ...]

    DeclareLaunchArgument(
      'expected_update_rate', default_value='15.0',
      description='Expected lidar frame rate. Ideally, set it slightly higher than actual frame rate, like 15 Hz for 10 Hz lidar scans.'),

    DeclareLaunchArgument(
      'assembling_time', default_value='1.0',
      description='How much time (sec) we assemble lidar scans before sending them to mapping node.'),

    DeclareLaunchArgument(
      'deskewing_slerp', default_value='true',
      description='Use fast slerp interpolation between first and last stamps of the scan for deskewing. It would less accruate than requesting TF for every points, but a lot faster. Enable this if the delay of the deskewed scan is significant larger than the original scan.'),

    DeclareLaunchArgument(
      'qos', default_value='1',
      description='Quality of Service: 0=system default, 1=reliable, 2=best effort.'),

    description_launch_group,
    OpaqueFunction(function=launch_setup),
  ])
```

Basic test procedure is: build a map using a ros bag, and than changing rtab to loclization mode and running a different rosbag on it. Note the two rosbags are recorded in the same location, with the bag used for mapping being the bigger of the two.

Thanks

Edwardius, 2025-06-05:
Here's an image of current results.

[image]

It seems that RTAB cannot relocalize on a previously made map with only Lidar

matlabbe, 2025-06-08:
RTAB-Map cannot globally re-localize with only a LiDAR. You have to send a 2d pose estimate under `/initialpose` topic to initialize the first pose, afterwards, RTAB-Map can locally localize. Here an example with [husky demo](https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_demos#clearpath-husky-nav2-3d-lidar-and-rgb-d-slam) with camera disabled. I launched the demo first in mapping mode, moved a little and then relaunched in localization mode. In the gif below, it is after relaunching in localization mode. If you add `/localization_pose` topic in RVIZ, the whole screen will be pink, because the localization covariance is set to 9999, meaning it is lost. By using 2D pose Estimate button, which sends `initialpose` topic, we can teleport the robot in the map. RTAB-Map still try to localize locally, so as soon we send an initial pose that is close to reality, RTAB-Map will snap back at the right position in the map, shrinking the localization pose's covariance.

[image]

Edwardius, 2025-06-08:
Hi @matlabbe, thanks for the clarification! I suspected so the more I dug deeper into the code.

As an addition, I was wondering about apriltag usage in RTAB. It seems that if we want to use the functionality, then a complete recompile of the code is needed. Is there docs/samples pointing towards using apriltags? Is global relocalization possible if an apriltag is detected?

matlabbe, 2025-06-09:
We can build `rtabmap_slam` with `apriltag_msgs` support, then we could have an external node publishing on `tag_detections` (type `apriltag_msgs/msg/AprilTagDetectionArray`) input topic of `rtabmap` node.

Other option is to rebuild rtabmap library with OpenCV+ArUco module, then set parameter `RGBD/MarkerDetection` to "true" and the corresponding parameters. Except the first four parameters, all the others also affect external tag detection (`tag_detections` topic).
```
rtabmap --params | grep "Param: Marker/"
Param: Marker/CornerRefinementMethod = "0"                 [Corner refinement method (0: None, 1: Subpixel, 2:contour, 3: AprilTag2). For OpenCV <3.3.0, this is "doCornerRefinement" parameter: set 0 for false and 1 for true.]
Param: Marker/Dictionary = "0"                             [Dictionary to use: DICT_ARUCO_4X4_50=0, DICT_ARUCO_4X4_100=1, DICT_ARUCO_4X4_250=2, DICT_ARUCO_4X4_1000=3, DICT_ARUCO_5X5_50=4, DICT_ARUCO_5X5_100=5, DICT_ARUCO_5X5_250=6, DICT_ARUCO_5X5_1000=7, DICT_ARUCO_6X6_50=8, DICT_ARUCO_6X6_100=9, DICT_ARUCO_6X6_250=10, DICT_ARUCO_6X6_1000=11, DICT_ARUCO_7X7_50=12, DICT_ARUCO_7X7_100=13, DICT_ARUCO_7X7_250=14, DICT_ARUCO_7X7_1000=15, DICT_ARUCO_ORIGINAL = 16, DICT_APRILTAG_16h5=17, DICT_APRILTAG_25h9=18, DICT_APRILTAG_36h10=19, DICT_APRILTAG_36h11=20]
Param: Marker/Length = "0"                                 [The length (m) of the markers' side. 0 means automatic marker length estimation using the depth image (the camera should look at the marker perpendicularly for initialization).]
Param: Marker/MaxDepthError = "0.01"                       [Maximum depth error between all corners of a marker when estimating the marker length (when Marker/Length is 0). The smaller it is, the more perpendicular the camera should be toward the marker to initialize the length.]
Param: Marker/MaxRange = "0.0"                             [Maximum range in which markers will be detected. <=0 for unlimited range.]
Param: Marker/MinRange = "0.0"                             [Miniminum range in which markers will be detected. <=0 for unlimited range.]
Param: Marker/Priors = ""                                  [World prior locations of the markers. The map will be transformed in marker's world frame when a tag is detected. Format is the marker's ID followed by its position (angles in rad), markers are separated by vertical line ("id1 x y z roll pitch yaw|id2 x y z roll pitch yaw"). Example:  "1 0 0 1 0 0 0|2 1 0 1 0 0 1.57" (marker 2 is 1 meter forward than marker 1 with 90 deg yaw rotation).]
Param: Marker/PriorsVarianceAngular = "0.001"              [Angular variance to set on marker priors.]
Param: Marker/PriorsVarianceLinear = "0.001"               [Linear variance to set on marker priors.]
Param: Marker/VarianceAngular = "0.01"                     [Angular variance to set on marker detections. If Marker/VarianceOrientationIgnored is enabled, it is ignored with Optimizer/Strategy=1 (g2o) and it corresponds to bearing variance with Optimizer/Strategy=2 (GTSAM).]
Param: Marker/VarianceLinear = "0.001"                     [Linear variance to set on marker detections. If Marker/VarianceOrientationIgnored is enabled and Optimizer/Strategy=2 (GTSAM): it is the variance of the range factor, with 9999 to disable range factor and to do only bearing.]
Param: Marker/VarianceOrientationIgnored = "false"         [When this setting is false, the landmark's orientation is optimized during graph optimization. When this setting is true, only the position of the landmark is optimized. This can be useful when the landmark's orientation estimation is not reliable. Note that for Optimizer/Strategy=1 (g2o), only Marker/VarianceLinear needs be set if we ignore orientation. For Optimizer/Strategy=2 (GTSAM), instead of optimizing the landmark's position directly, a bearing/range factor is used, with Marker/VarianceLinear as the variance of the range factor (with 9999 to optimize the position with only a bearing factor) and Marker/VarianceAngular as the variance of the bearing factor (pitch/yaw).]
```

Yes, global localization can work with apriltags/markers, or any sensor that can provide a relative pose on a topic of type `apriltag_msgs/msg/AprilTagDetectionArray`.

I am currently preparing a new ROS2 binary release (Humble/Jazzy/Kilted), I'll include `apriltag_msgs` as dependency already to avoid rebuilding the package for the `tag_detections` topic support.

Edwardius, 2025-06-09:
Awesome, thanks for the quick support :)

closed 2025-06-09 by Edwardius

Edwardius, 2025-06-09:
Oh, one more thing. Is there a mechanism in place to trigger a request for a new initial pose? ie. localization covariance reaches some threshold, and so rtab notifies so through some topic, and we will have to pass a pose into some other topic to localize again.

Edwardius, 2025-06-09:
Also, is there a roadmap for integrating some sort of global localization mechanism for lidar? IE. particle filter or https://arxiv.org/pdf/2310.10023

I'd be happy to help :)

harsh-saini-prd, 2026-10-01:
@matlabbe I was trying to see whether there is a way we can define a localisation score for when the robot is manoeuvring in a map. I identified some parameter and topic starting from the Hypothesis_value under the stat_key and value in /rtabmap/info, then the visual_inliers, visual_inlier_ratio and the covariance of localization_pose.
can you point me whether this approach is good or should i focus on the loc_thr. But i think simply depending on it would be one sided because there are cases where the covariance would go high beyond a threshold but still the robot is localised so a perfect threshold cannot be set. 
I have been working with rtab for a while now but still new to it so a input would be of help.

matlabbe, 2026-10-18:
In general, localization_pose covariance would be a good indication to know how confident localization is in the map. A large covariance means the robot didn't re-localize for a while, so more chance to be off the actual real localization in the map, though not necessary lost in all cases. To discuss more about localization performance in general and not related to this thread, I recommend to start a new thread.