introlab/rtabmap_ros #1326: rtabmap not loading

NamaiBest, 2025-06-11:
Hi there,

I'm fairly new to using RTAB-Map, and I could use some help getting started. Here's my current setup:

    Device: Jetson Orin

    OS: Ubuntu 22.04

    ROS: ROS 2 Humble

    Camera: Intel RealSense D435i

I’ve successfully installed both `rtabmap` and `rtabmap_ros`. However, I’m unsure how to proceed with mapping a room using this setup. I understand that multiple nodes might need to be launched, possibly in separate terminal windows, but the available documentation for ROS 2 is quite limited.

Specifically, I want to generate and visualize the map in RViz so I can see it in real time and understand how well the mapping is progressing.

Could you please guide me step-by-step on how to launch the necessary nodes and map the environment correctly?

Thanks in advance!

matlabbe, 2025-06-13:
`realsense_d400.launch.py` doesn't provide `rviz` option. You may launch rviz like this:
```
ros2 run rviz2 rviz2 -d $(ros2 pkg prefix rtabmap_launch)/share/rtabmap_launch/launch/config/rgbd.rviz
```
and adjust the topics inside the display (remove `/rtabmap` namespace). You can also copy that file somewhere else, but you could also just use your default rviz config and re-add the same displays (those under `rtabmap_rviz_plugins` group in rviz). for the occupancy grid topic, use the standard display `Map` for it, and adjust the topic.

matlabbe, 2025-06-17:
We created `realsense_d400.launch.py` example because the tweaks needed cannot be done with `rtabmap.launch.py`. In particular, if you have a D435i, I recommend to use [realsense_d435i_infra.launch.py](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_infra.launch.py) for best accuracy. In that launch, we:
1. Use left IR + Depth (less motion blur than RGB camera, and larger FOV, and perfect depth registration and synchronization)
2. Disable IR emitter (required when IR is used for VSLAM)
3. Enable IMU (align map with gravity)

Note that `rtabmap.launch.py` is here just for quick testing (and to match some ros1 examples using `rtabmap.launch`, see [RGB-D](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) and [Stereo](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping) tutorials), but I recommend to make your own launch files similar to ros2 examples we provide here: https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_examples/launch, and if you need to integrate on a robot, see the demos: https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_demos#rtabmap_demos

matlabbe, 2025-06-18:
```
failed to claim usb interface, interface 0, is busy - retrying...
```
Do not do `ros2 launch realsense2_camera rs_launch.py`, the realsense driver is launched from [realsense_d435i_infra.launch.py](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_infra.launch.py):
 https://github.com/introlab/rtabmap_ros/blob/d336369ca4ac4f979ed2836658137acf3b3677c2/rtabmap_examples/launch/realsense_d435i_infra.launch.py#L43-L54

So you are spawning another realsense2 driver which is in conflicts with the other one.

You can also do `ros2 run rviz2 rviz2 -d $(ros2 pkg prefix rtabmap_launch)/share/rtabmap_launch/launch/config/rgbd.rviz` if you want rviz2. `rtabmap_viz` can be closed if you don't need it.