introlab/rtabmap_ros #1341: Why Is Only a Partial Point Cloud Shown During RTAB-Map Localization?

pisquareroot, 2025-07-10:
Hi,

I'm currently working with RTAB-Map for localization, but I'm encountering an issue where only a partial point cloud is being visualized. I'm unsure why this is happening and would appreciate any insight into potential causes.

[image]

Additionally, I already have a saved .db map file from a previous mapping session, but I'm not entirely sure how to properly use it to resume localization, continue mapping, or perform navigation. If you have any relevant resources, examples, or suggestions on how to handle this workflow, I would be incredibly grateful.

matlabbe, 2025-07-11:
Can you share the database? A partial map like this maybe be caused by many issues.

matlabbe, 2025-07-14:
3D occupancy visualization would be handled by rviz2, with octomap rviz plugins and subscribing to generated octomap from rtabmap node (`Grid/3D` should be enabled to generated the OctoMap).

For the elevation map, you will need to rebuild rtabmap/rtabmap_ros from source with [grid_map_core](https://github.com/ANYbotics/grid_map) already installed.

For the database, the covariance of the odometry looks like this:
```
[1e-05, 0, 0, 0, 0, 0;
 0, 1e-05, 0, 0, 0, 0;
 0, 0, 1000000000000, 0, 0, 0;
 0, 0, 0, 1000000000000, 0, 0;
 0, 0, 0, 0, 1000000000000, 0;
 0, 0, 0, 0, 0, 0.001]
```
and `Reg/Force3DoF` parameter is false, so the optimizer is failing to optimize the map. Setting `Reg/Force3DoF=true` fixed the graph optimization issue:

[image]

Note that if you expect to work in 3D instead of 2D, you may have to put reasonable variance to Z, roll and pitch parameters of the covariance matrix.

matlabbe, 2025-07-15:
Either 3D or 2D, continuing mapping a previously built map is done the same way. Basically, you just need to restart rtabmap node without `--delete_db_on_start` argument, it will start a new session in the same database. When a loop closure is detected with the previous map in the database, the output occupancy grid will be updated with the old map, showing both maps in the occupancy grid. See this example: https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_demos#multi-session-indoor-2d-lidar-and-rgb-d-slam and [video](https://youtu.be/XrnyhaxPCro).