introlab/rtabmap_ros #1342: UserWarning: Unknown distribution option: 'tests_require'

Plumezz, 2025-07-13:
- ROS Version: Humble

- OS Version: Ubuntu22.04

- Python Version: 3.10.12

- setuptools Version: 75.2.0

Hello, when I build this project from source, some warnings about package `rtabmap_python` occurred.
Hers is the output of the command colcon build:
```
Starting >>> rtabmap_python
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
--- stderr: rtabmap_python                                        
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
---
Finished <<< rtabmap_python [1.29s]

```

This warning shows the `tests_require` option is expired. But the python and setuptools are default package with the ubuntu22.04.

And then I use `colcon test` to test this project, there were also some warnings.

Here is the output about the command `colcon test`:
```
Starting >>> rtabmap_python
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
Finished <<< rtabmap_msgs [0.32s]                           
Starting >>> rtabmap_conversions
--- stderr: rtabmap_python

----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
---
Finished <<< rtabmap_python [0.32s]
```

It seems like that there were no python testing programs to run and no result file `pytest.xml` was generated in build directory.

Then I tried to modify the `setup.py` file in package `rtabmap_python` and replaced the option `tests_require` with `extras_require`.

Finally these packages can be built correctly.

Here is the modified content in `setpu.py` file:
```
tests_require=['pytest'],  # remove
extras_require={
        'test': ['pytest'],
 },  # add
```
And then I run `colcon test` command, some errors about `test_pep257` occurred.

Here is the content of `pytest.xml` file of `rtabmap_python` package.
```
<?xml version="1.0" encoding="utf-8"?><testsuites><testsuite name="pytest" errors="0" failures="2" skipped="0" tests="3" time="0.770" timestamp="2025-07-13T16:43:08.853849+08:00" hostname="qiUbuntu"><testcase classname="rtabmap_python.test.test_copyright" name="test_copyright" time="0.165"><failure message="AssertionError: Found errors&#10;assert 1 == 0">test/test_copyright.py:23: in test_copyright
    assert rc == 0, 'Found errors'
E   AssertionError: Found errors
E   assert 1 == 0</failure></testcase><testcase classname="rtabmap_python.test.test_flake8" name="test_flake8" time="0.267"><failure message="AssertionError: Found 10 code style errors / warnings:&#10;  ./setup.py:18:17: Q003 Change outer quotes to avoid escaping inner quotes&#10;  ./rtabmap_python/compression.py:3:1: I100 Import statements are in the wrong order. 'import struct' should be before 'import zlib'&#10;  ./rtabmap_python/compression.py:4:1: I201 Missing newline between import groups. 'import numpy' is identified as Third Party and 'import struct' is identified as Stdlib.&#10;  ./rtabmap_python/compression.py:6:1: E302 expected 2 blank lines, found 1&#10;  ./rtabmap_python/compression.py:22:40: Q000 Double quotes found but single quotes preferred&#10;  ./rtabmap_python/compression.py:26:1: E302 expected 2 blank lines, found 1&#10;  ./rtabmap_python/compression.py:26:16: A002 argument &quot;bytes&quot; is shadowing a python builtin&#10;  ./rtabmap_python/compression.py:31:47: Q000 Double quotes found but single quotes preferred&#10;  ./rtabmap_python/compression.py:34:1: W391 blank line at end of file&#10;  ./rtabmap_python/__init__.py:2:1: W391 blank line at end of file&#10;assert 1 == 0">test/test_flake8.py:23: in test_flake8
    assert rc == 0, \
E   AssertionError: Found 10 code style errors / warnings:
E     ./setup.py:18:17: Q003 Change outer quotes to avoid escaping inner quotes
E     ./rtabmap_python/compression.py:3:1: I100 Import statements are in the wrong order. 'import struct' should be before 'import zlib'
E     ./rtabmap_python/compression.py:4:1: I201 Missing newline between import groups. 'import numpy' is identified as Third Party and 'import struct' is identified as Stdlib.
E     ./rtabmap_python/compression.py:6:1: E302 expected 2 blank lines, found 1
E     ./rtabmap_python/compression.py:22:40: Q000 Double quotes found but single quotes preferred
E     ./rtabmap_python/compression.py:26:1: E302 expected 2 blank lines, found 1
E     ./rtabmap_python/compression.py:26:16: A002 argument "bytes" is shadowing a python builtin
E     ./rtabmap_python/compression.py:31:47: Q000 Double quotes found but single quotes preferred
E     ./rtabmap_python/compression.py:34:1: W391 blank line at end of file
E     ./rtabmap_python/__init__.py:2:1: W391 blank line at end of file
E   assert 1 == 0</failure></testcase><testcase classname="rtabmap_python.test.test_pep257" name="test_pep257" time="0.012" /></testsuite></testsuites>
```
Now the python and setuptools version are updated very quickly, could you tell me if it is necessary to modify replace the `tests_require` option to adapt to the current environment.

And for the formatting errors detected by `test_pep257`, could you tell me how do you think about that? Is it necessary to fix that?

Thank you so much!

matlabbe, 2025-07-14:
We don't have any tests, so it makes sense it "ran 0 tests". For this line:
https://github.com/introlab/rtabmap_ros/blob/7d3b5244146e02bf2228f964d5a46a5285ae65e4/rtabmap_python/setup.py#L20
I think it was coming from the ros2 python package generator, we could remove that line.

Plumezz, 2025-07-17:
OK, thanks for your reply.

closed 2025-07-18 by matlabbe