introlab/rtabmap_ros #1360: Timestamp alignment issue

lifafang, 2025-09-22:
I'm debugging with an Orbbec camera and encountered this issue. How can I resolve it

“”“
[rgbd_odometry-2] [WARN] [1758513759.089787811] [rtabmap.rgbd_odometry]: Dropping image/scan data with stamp 1758513758.977694 (delay=0.112090). Something is wrong because the clock difference with the previous topic received (0.020811s) is much lower than the expected one (0.033184s) estimated from the topic stamps (previous stamp=1758513758.944510). If you are processing a large bag with flaky replaying delay, consider setting parameter "always_process_most_recent_frame:=false" to avoid aggressively dropping data.
[rgbd_odometry-2] [WARN] [1758513759.114679726] [rtabmap.rgbd_odometry]: Dropping image/scan data with stamp 1758513759.010879 (delay=0.103798). Something is wrong because the clock difference with the previous topic received (0.024894s) is much lower than the expected one (0.033185s) estimated from the topic stamps (previous stamp=1758513758.977694). If you are processing a large bag with flaky replaying delay, consider setting parameter "always_process_most_recent_frame:=false" to avoid aggressively dropping data.
”“”

matlabbe, 2025-09-28:
I [updated](https://github.com/introlab/rtabmap_ros/commit/f9ad71a7afda843011dae31cc80f82e0ecc034ce) the code recently to be less sensible to that warning. In your cases, the warning would not have appear. However, the frame is still dropped.

[image]

What is detected here is that images should be published at 33 Hz or 33 ms period, but we received two consecutive frames with time difference below 33 ms, which should not be possible if the sensor has a fixed capture rate. Without `always_process_most_recent_frame:=false`, odometry will aggressively drop frames: if odometry is currently processing an image and receives a new one, the new one is dropped. So if the time difference is under the expected one, it means maybe the camera driver is giving wrong timestamps if the camera is really capturing at 30 Hz. Thus if odometry could be computed in 30 ms, so normally not skipping any frames, but then receive a frame under 20 ms even if in reality images are captured at 30 Hz, it will skip it wrongly.

lifafang, 2025-10-01:
If I want to use SuperPoint for feature extraction in a ROS 2 environment, how should I proceed? Could you provide some reference examples? @matlabbe

matlabbe, 2025-10-02:
This Dockerfile can be a good starting point: https://github.com/introlab/rtabmap_ros/blob/master/docker/noetic/superpoint/Dockerfile, by installing ROS2 Humble instead of ROS Noetic. I'll check later if I could make a dockerfile for humble.