introlab/rtabmap_ros #1363: Cannot use rgb and rgb-d together for slam

KimchiPizza, 2025-09-24:
I want to proceed slam using the rgb and rgbd cameras. So
I tried to run the subscribe_rgb and subscribe_rgbd parameter set to true, 
but it failed to run.

_[rtabmap-2] [WARN] [1758615719.881961864] [rtabmap]: rtabmap: Parameters subscribe_rgb and subscribe_rgbd cannot be true at the same time. Parameter subscribe_rgb is set to false._

For reference, the RGB camera is a 180-degree wide-angle camera. I was able to run SLAM with the RGB camera alone by SfM parameters. I really want to run SLAM using RGB and RGB-D simultaneously.

Is there a better approach? Any advice would be greatly appreciated.

thank you.

matlabbe, 2025-09-25:
> I was able to run SLAM with the RGB camera alone by SfM parameters.

With rtabmap_ros?

To feed RGB and RGB-D data at the same time, you will need to use `rgb_sync` and `rgbd_sync` nodes, then synchronize their outputs (`rgbd_image`) with either:
1. `rgbdx_sync` and use `subscribe_rgbd: true` and `rgbd_cameras: 0` for rtabmap node, then connect `rgbd_images` topic.
2. use `subscribe_rgbd: true` and `rgbd_cameras: 2` for rtabmap node, then remap `rgbd_image0` to topic from `rgb_sync` and `rgbd_image1` to topic from `rgbd_sync`.

Note however that if the cameras cannot be hardware synchronized, they should have at least 30 Hz or more so that the approximate synchronization is not too bad, or move slowly.

matlabbe, 2025-10-16:
Make sure all camera info and image topics have a `frame_id` set in their header. The tf error says there is an empty `frame_id` somewhere.

KimchiPizza, 2025-10-17:
@matlabbe  Thank you for reply
I have two rgbd_sync nodes, and one rtabmap node. there's no odometry node.
Since there is no odometry node, the odom frame id parameter has not been set until this time.

This is rtabmap node's parameters.
 ```
   parameters={
          'frame_id':'base_footprint',
          'use_sim_time':use_sim_time,
          'rgbd_cameras': 2,            
          'subscribe_rgbd':True,
          'subscribe_rgb':False,      
          'subscribe_scan':True,
          'subscribe_odom':False,
        #   'subscribe_odom_info':True,  
          'use_action_for_goal':True,
          'qos_scan':qos,
          'qos_image':qos,
          'qos_imu':qos,

          'Mem/StereoFromMotion':'False',
          'Reg/Strategy':'1',
          'Reg/Force3DoF':'true',
          'RGBD/NeighborLinkRefining':'True',
          'Grid/RangeMin':'0.15', 
          'Optimizer/GravitySigma':'0', 
          'approx_sync':True,
          'approx_sync_max_interval':0.3,
          'queue_size':50,
    }
```

Since you mentioned the odometry frame id, I set the odometry frame id parameter and ran.
`odom_frame_id':'odom',`

but ERROR occored
```
[INFO] [launch]: All log files can be found below /home/jetson/.ros/log/2025-10-17-12-47-27-434469-yahboom-499216
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgbd_sync-1]: process started with pid [499253]
...
[rtabmap-3] [INFO] [1760673736.301262294] [rtabmap]: rtabmap: qos_user_data   = 0
[rtabmap-3] [INFO] [1760673736.301272727] [rtabmap]: rtabmap: approx_sync     = true
[rtabmap-3] [INFO] [1760673736.301433915] [rtabmap]: Setup rgbd2 callback
[rtabmap-3] [INFO] [1760673736.316140227] [rtabmap]: 
[rtabmap-3] rtabmap subscribed to (approx sync):
[rtabmap-3]    /rgbd_image0 \
[rtabmap-3]    /rgbd_image1 \
[rtabmap-3]    /scan
[rtabmap-3] *** stack smashing detected ***: terminated
[ERROR] [rtabmap-3]: process has died [pid 377181, exit code -6, cmd '/home/jetson/rtabmap_ws/install/rtabmap_slam/lib/rtabmap_slam/rtabmap -d --ros-args --params-file /tmp/launch_params_8j2xfwab --params-file /tmp/launch_params_8bq_koaa'].
```

KimchiPizza, 2025-10-20:
@matlabbe 
Hello,

my rtabmap is 0.21.10-humble. And I hardcoded rtabmap 0.21.10 to add what you committed #3b5a4ed

At rgb_sync node, I set `fill_empty_depth` True.
At rtabmap node, 
```
  'subscribe_odom_info':False,
  'odom_frame_id':'base_footprint',
```

And This is my gdb log.

Thank you

```

#0  __pthread_kill_implementation
    (threadid=281472303482944, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at ./nptl/pthread_kill.c:44
#1  0x0000fffff667f244 in __pthread_kill_internal
    (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000fffff663a67c in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#3  0x0000fffff6627130 in __GI_abort () at ./stdlib/abort.c:79
#4  0x0000fffff6673300 in __libc_message
    (action=action@entry=do_abort, fmt=fmt@entry=0xfffff6752928 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:156
#5  0x0000fffff66f5a28 in __GI___fortify_fail
    (msg=msg@entry=0xfffff6752910 "stack smashing detected")
    at ./debug/fortify_fail.c:26
#6  0x0000fffff66f59f4 in __stack_chk_fail () at ./debug/stack_chk_fail.c:24
#7  0x0000fffff7d38700 in rtabmap_slam::CoreWrapper::commonMultiCameraCallbackImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rtabmap_msgs::msg::UserData_<std::allocator<void> > const> const&, std::vector<std::shared_ptr<cv_bridge::CvImage const>, std::allocator<std::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<std::shared_ptr<cv_bridge::CvImage const>, std::allocator<std::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > > const&, std::vector<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > > const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > const&, sensor_msgs::msg::PointCloud2_<std::allocator<void> > const&, std::shared_ptr<rtabmap_msgs::msg::OdomInfo_<std::allocator<void> > const> const&, std::vector<rtabmap_msgs::msg::GlobalDescriptor_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::GlobalDescriptor_<std::allocator<void> > > > const&, std::vector<std::vector<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> > > >, std::allocator<std::vector<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> > > > > > const&, std::vector<std::vector<rtabmap_msgs::msg::Point3f_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::Point3f_<std::allocator<void> > > >, std::allocator<std::vector<rtabmap_msgs::msg::Point3f_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::Point3f_<std::allocator<void> > > > > > const&, std::vector<cv::Mat, std::allocator<cv::Mat> > const&) () at /home/jetson/rtabmap_ws/install/rtabmap_slam/lib/librtabmap_slam_plugins.so
#8  0x0000fffff7d39048 in rtabmap_slam::CoreWrapper::commonMultiCameraCallback(std::shared_ptr<nav_msgs::msg::Odometry_<std::allocator<void> > const> const&, std::shared_ptr<rtabmap_msgs::msg::UserData_<std::allocator<void> > const> const&, std::vector<std::shared_ptr<cv_bridge::CvImage const>, std::allocator<std::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<std::shared_ptr<cv_bridge::CvImage const>, std::allocator<std::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > > const&, std::vector<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > > const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > const&, sensor_msgs::msg::PointCloud2_<std::allocator<void> > const&, std::shared_ptr<rtabmap_msgs::msg::OdomInfo_<std::allocator<void> > const> const&, std::vector<rtabmap_msgs::msg::GlobalDescriptor_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::GlobalDescriptor_<std::allocator<void> > > > const&, std::vector<std::vector<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> > > >, std::allocator<std::vector<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> > > > > > const&, std::vector<std::vector<rtabmap_msgs::msg::Point3f_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::Point3f_<std::allocator<void> > > >, std::allocator<std::vector<rtabmap_msgs::msg::Point3f_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::Point3f_<std::allocator<void> > > > > > const&, std::vector<cv::Mat, std::allocator<cv::Mat> > const&) () at /home/jetson/rtabmap_ws/install/rtabmap_slam/lib/librtabmap_slam_plugins.so
#9  0x0000ffff00000000 in  ()
```

KimchiPizza, 2025-10-21:
I rebuilded rtabmap_slam package as debug mode and tried agian.

this is my gdb log. bt full 
```
#0  __pthread_kill_implementation
    (threadid=281473208399936, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at ./nptl/pthread_kill.c:44
        tid = 1389493
        ret = 0
        pd = 0xffff9699b840

                    old_mask = {__val = {281474778507404, 281474842463448, 281474976674200, 0, 5, 281474842462528, 281474976674056, 281474976674040, 524302793, 281474729880544, 281474801728808, 5, 8192231, 281474976673812, 281474976673816, 281474799383832}}
        ret = <optimized out>
#1  0x0000fffff433f244 in __pthread_kill_internal
    (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000fffff42fa67c in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
        ret = <optimized out>
#3  0x0000fffff42e7130 in __GI_abort () at ./stdlib/abort.c:79
        save_stage = 1

                  act = {__sigaction_handler = {sa_handler = 0xffffffff96e0, sa_sigaction = 0xffffffff96e0}, sa_mask = {__val = {281474976683616, 281474842462528, 1, 0, 281474842463448, 187647121162240, 281474842487496, 16, 1, 187649999016088, 187649999015640, 16, 4294967295, 187649999016088, 281474778353496, 281474842439728}}, sa_flags = 1, sa_restorer = 0xaaaaab889098}
        sigs = {__val = {32, 0, 5, 0, 187647121162241, 281474842462528, 281474976674112, 281474842304640, 281474842178000, 281474842462528, 1697136, 281473208399936, 281473208401920, 187649999015640, 187650006598048, 281474976683584}}
#4  0x0000fffff4333300 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0xfffff4412928 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:156
        ap = {__stack = 0xffffffff7270, __gr_top = 0xffffffff7270, __vr_top = 0xffffffff7240, __gr_offs = -40, __vr_offs = 0}
        fd = <optimized out>
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#5  0x0000fffff43b5a28 in __GI___fortify_fail (msg=msg@entry=0xfffff4412910 "stack smashing detected") at ./debug/fortify_fail.c:26
#6  0x0000fffff43b59f4 in __stack_chk_fail () at ./debug/stack_chk_fail.c:24
#7  0x0000fffff764d1f4 in rtabmap_slam::CoreWrapper::commonMultiCameraCallbackImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rtabmap_msgs::msg::UserData_<std::allocator<void> > const> const&, std::vector<std::shared_ptr<cv_bridge::CvImage const>, std::allocator<std::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<std::shared_ptr<cv_bridge::CvImage const>, std::allocator<std::shared_ptr<cv_bridge::CvImage const> > > const&, std::vector<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > > const&, std::vector<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, std::allocator<sensor_msgs::msg::CameraInfo_<std::allocator<void> > > > const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > const&, sensor_msgs::msg::PointCloud2_<std::allocator<void> > const&, std::shared_ptr<rtabmap_msgs::msg::OdomInfo_<std::allocator<void> > const> const&, std::vector<rtabmap_msgs::msg::GlobalDescriptor_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::GlobalDescriptor_<std::allocator<void> > > > const&, std::vector<std::vector<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> > > >, std::allocator<std::vector<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::KeyPoint_<std::allocator<void> > > > > > const&, std::vector<std::vector<rtabmap_msgs::msg::Point3f_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::Point3f_<std::allocator<void> > > >, std::allocator<std::vector<rtabmap_msgs::msg::Point3f_<std::allocator<void> >, std::allocator<rtabmap_msgs::msg::Point3f_<std::allocator<void> > > > > > const&, std::vector<cv::Mat, std::allocator<cv::Mat> > const&) (this=0xaaaaab8844a0, odomFrameId="base_footprint", userDataMsg=std::shared_ptr<const rtabmap_msgs::msg::UserData_<std::allocator<void> >> (empty) = {...}, imageMsgs=Python Exception <class 'gdb.error'>: There is no member named _M_impl.
, depthMsgs=Python Exception <class 'gdb.error'>: There is no member named _M_impl.
, cameraInfoMsgs=std::vector of length 2, capacity 2 = {...}, depthCameraInfoMsgs=std::vector of length 2, capacity 2 = {...}, scan2dMsg=..., scan3dMsg=..., odomInfoMsg=std::shared_ptr<const rtabmap_msgs::msg::OdomInfo_<std::allocator<void> >> (empty) = {...}, globalDescriptorMsgs=std::vector of length 0, capacity 0, localKeyPointsMsgs=std::vector of length 2, capacity 2 = {...}, localPoints3dMsgs=std::vector of length 2, capacity 2 = {...}, localDescriptorsMsgs=std::vector of length 2, capacity 2 = {...}) at /home/jetson/rtabmap_ws/rtabmap_ros/rtabmap_slam/src/CoreWrapper.cpp:1721
        timerConversion = {startTimeRecorded_ = {tv_sec = 1761036256, tv_usec = 835400}, stopTimeRecorded_ = {tv_sec = 1761036256, tv_usec = 835400}}
        rgb = {flags = 1124024336, dims = 2, rows = 0, cols = 0, data = 0x0, datastart = 0x0, dataend = 0x0, datalimit = 0x0, allocator = 0x0, u = 0x0, size = {p = 0xffffffff74b0}, step = {p = 0xffffffff74f8, buf = {3840, 3}}}
        depth = {flags = 1124024322, dims = 2, rows = 0, cols = 0, data = 0x0, datastart = 0x0, dataend = 0x0, datalimit = 0x0, allocator = 0x0, u = 0x0, size = {p = 0xffffffff7510}, step = {p = 0xffffffff7558, buf = {2560, 2}}} [... cut ...] reading variable: Cannot access memory at address 0x8>, localDescriptors=<error reading variable: Cannot access memory at address 0x8>) at /home/jetson/rtabmap_ws/rtabmap_ros/rtabmap_slam/src/CoreWrapper.cpp:1319
        lock = {mutex_ = @0xaaaaab888e40}
        odomFrameId = "base_footprint"
        __func__ = "commonMultiCameraCallback"
#9  0x0000ffff00000000 in  ()

   Warning: 'set logging off', an alias for the command 'set logging enabled', is deprecated.
Use 'set logging enabled off'.
```

closed 2025-11-01 by matlabbe